ros-drivers/velodyne #166: Compiling Error

FabianKuzaj, 2018-05-16:
Hi!

I started using ROS a few days ago and now want to integrate our VLP-16. So I created a catkin workspace first (as described here: http://wiki.ros.org/ROS/Tutorials/InstallingandConfiguringROSEnvironment) and then followed the instructions in this tutorial to install VLP-16 drivers: http://wiki.ros.org/velodyne/Tutorials/Getting%20Started%20with%20the%20Velodyne%20VLP16
But when it comes to building the workspace with catkin_make, the following error occurs:

    [ 58%] Built target velodyne_rawdata
    [ 64%] Built target ringcolors_nodelet
    [ 66%] Linking CXX executable /home/fabian/catkin_ws/devel/lib/velodyne_pointcloud/ringcolors_node
    Scanning dependencies of target cloud_node
    [ 68%] Building CXX object velodyne/velodyne_pointcloud/src/conversions/CMakeFiles/cloud_node.dir/cloud_node.cc.o
    CMakeFiles/velodyne_node.dir/velodyne_node.cc.o: In function `main':
    velodyne_node.cc:(.text+0x70): undefined reference to `ros::init(int&, char**, std::string const&, unsigned int)'

etc...

My first assumption was, that there are dependencies missing so I tried `$rosdep install --from-paths src --ignore-src --rosdistro kinetic -y` again (I'm using ros kinetic), but it didn't help. So I reset the entire catkin workspace and tried it again, but still the same problem.

Do you guys have any ideas?

Thanks,
Fabian

JWhitleyAStuff, 2018-05-18:
@FabianKuzaj - This looks like an environmental problem. Are you sourcing /opt/ros/kinetic/setup.bash in your ~/.bashrc file? Have you created a new terminal window since you installed ROS (this is what sources ~/.bashrc)?

ros::init is part of the roscpp package so try `sudo apt install ros-<ros_distro>-roscpp` (though this should already be installed if you used a -desktop variant of ROS).

JWhitleyAStuff, 2018-05-18:
Also worth noting that you can now install ros-<kinetic>-velodyne via `apt` and it will get version 1.3.0 (released November of last year) which has support for most sensors.

FabianKuzaj, 2018-05-18:
@JWhitleyAStuff Thanks for your reply.
I'm sourcing /opt/ros/kinetic/setup.bash as well as /home/fabian/catkin_ws/devel/setup.bash in my .bashrc. And I'm only using one terminal ($ROS_PACKAGE_PATH is /home/fabian/catkin_ws/src:/opt/ros/kinetic/share). I also tried to compile with just /opt/ros/kinetic/setup.bash in the .bashrc, didn't work..
`roscpp` is also installed.

Furthermore I installed the `ros-kinetic-velodyne` package via `apt`. But when I try to launch a package (for example `roslaunch velodyne_pointcloud VLP16_points.launch`) I get the following error:

    [ERROR] [1526684472.991399726]: Failed to load nodelet [/velodyne_nodelet_manager_driver] of type [velodyne_driver/DriverNodelet] even after refreshing the cache: Failed to load library /home/fabian/catkin_ws/devel/lib//libdriver_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = /home/fabian/catkin_ws/devel/lib//libdriver_nodelet.so: undefined symbol: _ZN12class_loader20class_loader_private30getCurrentlyLoadingLibraryNameEv)
    [ERROR] [1526684472.991434064]: The error before refreshing the cache was: Failed to load library /home/fabian/catkin_ws/devel/lib//libdriver_nodelet.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = /home/fabian/catkin_ws/devel/lib//libdriver_nodelet.so: undefined symbol: _ZN12class_loader20class_loader_private30getCurrentlyLoadingLibraryNameEv)
 etc..

JWhitleyAStuff, 2018-05-18:
Are you using the --extend flag on your second source line?

FabianKuzaj, 2018-05-19:
No, the last two lines of my .bashrc are (if this is what you mean):
...

    source /opt/ros/kinetic/setup.bash
    source /home/fabian/catkin_ws/devel/setup.bash

FabianKuzaj, 2018-05-19:
Yep, I cloned the git into the /src folder..
Maybe something in the ROS installation process went wrong? I'll try to reinstall everything (including ROS) on Monday, maybe that might help.

FabianKuzaj, 2018-05-21:
Reinstalled ROS, still getting the same compiling error. So if you have any ideas let me know. I now try to work with the raw UDP stream and convert it.

JWhitleyAStuff, 2018-05-21:
Are you using g++ as your compiler? What OS are you running on? Since this is during compiling and not linking, it looks like an issue with Include folders. These should be set by CMake in the variable ${PROJECT_NAME}_CATKIN_DEPS (line 4 of CMakeLists.txt in `velodyne_pointcloud`) which would point it to /opt/ros/kinetic/include where it should find ros/ros.h which points to ros/init.h (where the ros::init function is found). Try to trace it back through this path and see if it helps any.

FabianKuzaj, 2018-05-21:
So I'm using g++ 4.9.3 on Ubuntu 16.04.
I checked all files you mentioned and everything is available.

Also I double-checked the thrown error, it looks like this:

    [ 50%] Linking CXX executable /home/fabian/catkin_ws/devel/lib/velodyne_driver/velodyne_node
    [ 52%] Built target driver_nodelet
    [ 58%] Built target velodyne_rawdata
    [ 64%] Built target ringcolors_nodelet
    [ 66%] Linking CXX executable /home/fabian/catkin_ws/devel/lib/velodyne_pointcloud/ringcolors_node
    [ 68%] Linking CXX executable /home/fabian/catkin_ws/devel/lib/velodyne_pointcloud/cloud_node
    CMakeFiles/velodyne_laserscan_node.dir/src/node.cpp.o: In function `main':
    node.cpp:(.text+0x70): undefined reference to `ros::init(int&, char**, std::string const&, unsigned int)'
    node.cpp:(.text+0xcc): undefined reference to `ros::NodeHandle::NodeHandle(std::string const&, std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&)'
    node.cpp:(.text+0x143): undefined reference to `ros::NodeHandle::NodeHandle(std::string const&, std::map<std::string, std::string, std::less<std::string>, std::allocator<std::pair<std::string const, std::string> > > const&)'
    /home/fabian/catkin_ws/devel/lib/libvelodyne_laserscan.so: undefined reference to `class_loader::class_loader_private::getCurrentlyLoadingLibraryName()'

So seems like a linking error this time? However, I don't really understand why it works for the `velodyne_driver` folder, but not for the `velodyne_pointcloud` folder.

trainman419, 2018-05-23:
Cross-posting from https://answers.ros.org/question/291910/linking-problem-with-catkin_libraries/?answer=291949#post-id-291949 , I strongly recommend that you try using the default version of gcc instead of an older version.

FabianKuzaj, 2018-05-23:
@JWhitleyAStuff @trainman419 thanks again for your help! As mentioned above, problem was solved here: https://answers.ros.org/question/291910/linking-problem-with-catkin_libraries/
Issue can be closed.

closed 2018-05-23 by FabianKuzaj