ros-drivers/velodyne #171: Migrating to ROS Melodic

Alby407, 2018-06-11:
Is there already a date when the package will be migrated/updated to ROS Melodic?

JWhitleyAStuff, 2018-06-11:
There is no particular date but I'm working on adding Melodic to the list of test builds right now. After it is stable in Melodic and I've pulled the two PRs that need testing (#164 and #155), I'll make a new release for Kinetic, Lunar, and Melodic.

JWhitleyAStuff, 2018-06-11:
It is now building on Travis for Indigo, Kinetic, Lunar, and Melodic. As I said, I want to test changes on the two PRs and get them merged before releasing but I'll update this ticket when the new version is out.

Alby407, 2018-06-11:
Thank you very much!

JWhitleyAStuff, 2018-09-19:
Sorry this took so incredibly long. I just released v1.4.0 which is being built on the ROS build farm for Indigo, Kinetic, Lunar, and Melodic. I didn't get all of the PRs in that I wanted but a release was really necessary.

closed 2018-09-19 by JWhitleyAStuff