ros-drivers/velodyne #176: question about classify points between strongest return and last return

ttomchy, 2018-07-21:
Hello ,In the velodyne dataset ,They have provided three modes to return the data, the  strongest return and last return or both returns.  We can also set the return models on the site but how to classify the points belongs to the   strongest return and the last return when we using the dual mode to return data. I am confused about it. Does anyone know how to solve it ?  Thank you.

JWhitleyAStuff, 2018-07-24:
On the Velodynes, each laser emits two pulses per point. For a description of the difference between these three modes, I recommend seeing this document: http://velodynelidar.com/docs/notes/63-9277%20Rev%20B%20HDL-32E%20Application%20Note%20-%20Packet%20Structure%20&%20Timing%20Definition.pdf

When both returns are provided (Dual Return mode), you would need to look at the distance of each point to determine "last return" and the amplitude of each point to determine "strongest return." When you set the Velodyne to either of these modes, this is done automatically in the hardware. When you set the Velodyne to Dual Return mode, both returns are provided and you can determine in software which fits into these categories. Please let me know if this answers your question.

closed 2018-08-03 by JWhitleyAStuff

nivir, 2021-05-31:
@JWhitleyWork, the link is not working. Could you please update the link? Thank you!

JWhitleyWork, 2021-05-31:
@nivir Unfortunately, this was almost 3 years ago and Velodyne has changed their website significantly since then. You'll have to go to their site and download the users' guide for your sensor in order to find the details about the return types.

eugeniu1994, 2021-08-24:
Hello @nivir , @JWhitleyWork , @ttomchy 

I am using a Velodyne alpha prime lidar, and got stuck in splitting the point cloud by strong and late return. How to check more precisely, if the point is from the strong or late return. You mentioned looking at distance and amplitude, but what is the condition?
I would appreciate your hint on this

Thanks

alvariseppanen, 2022-10-31:
> @JWhitleyWork Thanks for the reply. I realize it's not a simple task, however, I managed to solve this problem for VLP16 lidar, and now I'm trying to modify the driver to provide dual return support for VLS 128 lidar as well.
> 
> I've read the provided datasheet, but I can't find any hints on how to check if the point is from strong or late return. The approach used for VLP 16 does not work for VLS 128.
> 
> Do you have any idea where I could get some extra information other than provided in the VLS128 datasheet?
> 
> Thanks

Hello,

I would like to know how did you solve this problem for vlp16 lidar? 

Thank you