ros-drivers/velodyne #180: velodyne LiDAR HDL 64E S2 wrong measurement of points

ajay1606, 2018-08-10:
Hello Every one,
I am currently using a Velodyne HDL 64 E S2 model of the sensor with Velodyne ros package. Any how we are getting data and able to render in rviz using the launch file **64e_s2_points** with calibration file **hdl64e_s2_calib.yaml**

But the problem is , projected points appears to be wrong. It is not clearly aligned with each channel data. 

// Launch file 64e_s2_points.launch
```
 <launch>
    <!-- pointcloud cloud_nodelet -->
      <arg name="velodyne_calibration" default="$(find velodyne_pointcloud)/params/hdl64e_s2_calib.yaml" />
      <!-- <arg name="velodyne_calibration" default="" /> -->
      <arg name="min_range" default="2.0" />
      <arg name="max_range" default="65.0" />
      <node pkg="nodelet" type="nodelet" name="cloud_nodelet"
            args="load velodyne_pointcloud/CloudNodelet velodyne_nodelet_manager">
        <param name="calibration" value="$(arg velodyne_calibration)"/>
        <param name="min_range" value="$(arg min_range)"/>
    <param name="max_range" value="$(arg max_range)"/>
  </node>

  <!-- start nodelet manager and load driver nodelet -->
  <node pkg="nodelet" type="nodelet" name="velodyne_nodelet_manager"
        args="manager" />
  <arg name="model" default="64E_S2" />
  <arg name="pcap" default="" />
  <arg name="read_once" default="true" />
  <arg name="read_fast" default="true" />
  <arg name="repeat_delay" default="0.0" />
      <!-- rotating parameter -->
      <!-- 1) if your setting rpm is 1200(20Hz) -->
      <!-- <arg name="rpm" default="850.0" /> -->
      <!-- 2) if your setting rpm is 900(15Hz) -->
      <!-- <arg name="rpm" default="850.0" /> -->
  <!-- 3) if your setting rpm is 600(10Hz) -->

  <arg name="rpm" default="570.0" />

  <!-- 4) if your setting rpm is 300(5Hz) -->
  <!-- <arg name="rpm" default="290.0" /> -->

  <arg name="frame_id" default="velodyne" />

      <node pkg="nodelet" type="nodelet" name="driver_nodelet"
            args="load velodyne_driver/DriverNodelet velodyne_nodelet_manager" >
        <param name="model" value="$(arg model)"/>
        <param name="pcap" value="$(arg pcap)"/>
        <param name="read_once" value="$(arg read_once)"/>
        <param name="read_fast" value="$(arg read_fast)"/>
        <param name="repeat_delay" value="$(arg repeat_delay)"/>
        <param name="rpm" value="$(arg rpm)"/>
        <param name="frame_id" value="$(arg frame_id)"/>
      </node>
</launch>
```
Calibration file: hdl64e_s2_calib.yaml
```
lasers:
- {dist_correction: 0.981637605, dist_correction_x: 0, dist_correction_y: 0, focal_distance: 0,
  focal_slope: 0, horiz_offset_correction: 0.025999999, laser_id: 0, max_intensity: 255,
  min_intensity: 0, rot_correction: -0.09709912400667693, vert_correction: -0.12678532453409336,
  vert_offset_correction: 0.19740748}
- {dist_correction: 1.2924338, dist_correction_x: 0, dist_correction_y: 0, focal_distance: 0,
  focal_slope: 0, horiz_offset_correction: -0.025999999, laser_id: 1, max_intensity: 255,
  min_intensity: 0, rot_correction: -0.06057167088908498, vert_correction: -0.12094779159811801,
  vert_offset_correction: 0.19783373}
- {dist_correction: 1.270309331, dist_correction_x: 0, dist_correction_y: 0, focal_distance: 0,
  focal_slope: 0, horiz_offset_correction: 0.025999999, laser_id: 2, max_intensity: 255,
  min_intensity: 0, rot_correction: 0.0384232803656565, vert_correction: 0.003765600221055326,
  vert_offset_correction: 0.20684389}
- {dist_correction: 1.2513616650000001, dist_correction_x: 0, dist_correction_y: 0,
[... 401 lines cut ...]
  min_intensity: 0, rot_correction: -0.16462279909895197, vert_correction: -0.23400086557751998,
  vert_offset_correction: 0.11933828}
- {dist_correction: 0.950396769, dist_correction_x: 0, dist_correction_y: 0, focal_distance: 0,
  focal_slope: 0, horiz_offset_correction: -0.025999999, laser_id: 59, max_intensity: 255,
  min_intensity: 0, rot_correction: -0.10591289361224963, vert_correction: -0.22780218911272196,
  vert_offset_correction: 0.11974475}
- {dist_correction: 1.13498927, dist_correction_x: 0, dist_correction_y: 0, focal_distance: 0,
  focal_slope: 0, horiz_offset_correction: 0.025999999, laser_id: 60, max_intensity: 255,
  min_intensity: 0, rot_correction: -0.027951906992282463, vert_correction: -0.16669180664204566,
  vert_offset_correction: 0.1236951}
- {dist_correction: 0.9873951700000001, dist_correction_x: 0, dist_correction_y: 0,
  focal_distance: 0, focal_slope: 0, horiz_offset_correction: -0.025999999, laser_id: 61,
  max_intensity: 255, min_intensity: 0, rot_correction: 0.02763049064789369, vert_correction: -0.15753373244427354,
  vert_offset_correction: 0.12427939}
- {dist_correction: 1.112591409, dist_correction_x: 0, dist_correction_y: 0, focal_distance: 0,
  focal_slope: 0, horiz_offset_correction: 0.025999999, laser_id: 62, max_intensity: 255,
  min_intensity: 0, rot_correction: -0.04956258775311027, vert_correction: -0.22002877187701952,
  vert_offset_correction: 0.12025283}
- {dist_correction: 0.951754713, dist_correction_x: 0, dist_correction_y: 0, focal_distance: 0,
  focal_slope: 0, horiz_offset_correction: -0.025999999, laser_id: 63, max_intensity: 255,
  min_intensity: 0, rot_correction: 0.008628967163054533, vert_correction: -0.21183750536509716,
  vert_offset_correction: 0.12078632000000002}
num_lasers: 64
```
[image]

JWhitleyAStuff, 2018-08-10:
@ajay1606 - Is this the default calibration file that is provided with the driver or one that you generated from the .xml file provided by Velodyne for this sensor?

ajay1606, 2018-08-10:
Hello Everyone,

Finally, the problem got solved. I received a calibration file from the Velodyne technical support. Also, they mentioned that each unit of the Velodyne sensor comes with the separate calibration file.
[image]

JWhitleyAStuff, 2018-08-10:
Thanks for getting back to us!

closed 2018-08-10 by JWhitleyAStuff