ros-drivers/velodyne #181: Wrong dimensions of objects

ashnarayan13, 2018-08-16:
Hi, 

I have pcap recording on a VLP32c lidar. The size of a measured object is wrong by almost half in the scan data when viewed with rviz. Is there some scaling done for the points? The height of a standard pylon and the distance between pylons is nearly half the length that was physically measured. 
Any idea how to fix this?

Thanks!

ashnarayan13, 2018-08-17:
@JWhitleyAStuff - The maint/vlp_32c_support has proper scale. But the scan is not consistent and flickers. 
[image]

You can see the rotation of the LiDAR scan. This didn't happen in the master branch. Any ideas on how to fix it? This issue doesn't happen on veloview as well. 

Thanks!

JWhitleyAStuff, 2018-08-17:
@ashnarayan13 - The maint/vlp_32_c branch was several PRs behind master. I've updated it and merged. Please pull, rebuild, and try this branch again. Thanks.

ashnarayan13, 2018-08-20:
Hi, 
The issue is resolved! 
Thank you.

closed 2018-08-20 by ashnarayan13