ros-drivers/velodyne #183: Synchronise Velodyne Timestamp with ros::clock

ClementLeBihan, 2018-08-30:
Hi,

I'm working on using packet timestamp to compute precisely the transformation in map frame. But the "ros stamp" header is not really precise. I would better use the timestamp in the velodyne UDP packet. 
The timestamp in velodyne packet is on 4 bytes and representing the uS from the beginning of the new hour. But I can't understand why it's so differtent from the corresponding ros time header.

For example, for a velodyne packet, we have :
 - ROS STAMP HEADER : 1533734983,392073154 (08/08/2018 15h29m43s392ms)
- VELODYNE TIMESTAMP 2747198412 uS (2747s since the hour i.e 45m47s)

I don't understand why the velodyne timestamp (45m47s) don't correspond to the (29m43s) of the ros timestamp. Can the velodyne clock be synchronized with the ros clock ?

Best,
LE BIHAN Clément

JWhitleyAStuff, 2018-08-30:
Hi, @ClementLeBihan - thanks for asking this question! Currently the Velodyne sensors can be synchronized with an external GPS device (a Garmin GPS Puck) which has to be configured specifically for this purpose. You can order one from AutonomouStuff or directly from Velodyne. There is currently no way to push timing data from the ROS environment through the driver to the sensor. AFAICT, it looks like it doesn't accept UDP packets for configuration, just the web interface or, in the case of the HDL-64E, the RS-485 serial interface.

However, if you do have the GPS unit connected, it would be possible to modify the driver to receive the GPS data from the device and publish it to ROS. If the driver published a `sensor_msgs/TimeReference`, you can use the [ntpd_driver](http://wiki.ros.org/ntpd_driver) to synchronize the system clock. This has been suggested and seems to be implemented in #53 but I haven't had time to test it. I just updated the `gps_imu` branch if you wouldn't mind testing it on your device.

ClementLeBihan, 2018-08-31:
Hi @JWhitleyAStuff !

Thank you so much for your answer. I did know that the GPS can send PPS to the Velodyne to synchronize Time, but I thought that without any GPS the Velodyne was able to synchronize with PC clock from Ethernet connection.

Thank you for telling me about the ntpd_driver. I didn't know this, and I think that we'll use this solution. Let's keep in touch !

Best,
Clément

closed 2018-08-31 by JWhitleyAStuff

ClementLeBihan, 2018-09-04:
Sorry to re-open the issue, but I found in the VLP32-C documentation that it is possible to send NMEA message (and PPS) from a laptop. (7.4.4 GPS Connection Scenarios)

Will the Velodyne timestamp be synchronized with our laptop clock if we send "fake" NMEA message with the laptop timestamp from laptop serial port ? 

Best,
Clément

trainman419, 2018-09-06:
The VLP32-C (and the VLP16) require both a PPS and NMEA signal in order to synchronize time. If you do this correctly the VLP32-C web interface will show that PPS is locked and that the time is synchronized. Generating the NMEA and PPS signals from a Linux laptop is probably possible, but it's beyond the scope of this issue. Maybe look for a post on a Linux high-precision time forum or mailing list.

dejongyeong, 2020-06-08:
@ClementLeBihan Hi Clement, I came across the same issue as well and would like to know how did you resolve the issue of Velodyne VLP32C Timestamp Synchronization. Are you using an external GPS or with ros::clock? Thanks.

JINGJAN, 2022-03-08:
@dejongyeong Hi Dejong, I would like to know did you solve the Velodyne VLP32C time Synchronization problem, with the laptop clock? Thanks!

dejongyeong, 2022-03-16:
@JINGJAN unfortunately, I did not solve it and I think you would have to incorporate with an external GPS for that. What I did was using the timestamps attached in the message header (from ROS)