ros-drivers/velodyne #184: Nodelet crashing

lstelzner, 2018-09-12:
Hi,
I installed ros-kinetic-velodyne, after sourcing my environment I run:
'roslaunch velodyne_pointcloud VLP16_points.launch'
Then it crashes:
~ $ roslaunch velodyne_pointcloud VLP16_points.launch 
... logging to /home/laura/.ros/log/fa4bc318-b63d-11e8-a5fd-34e12d206854/roslaunch-X1C-LStelzner-10961.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://10.0.0.113:44117/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.13
 * /velodyne_nodelet_manager_cloud/calibration: /opt/ros/kinetic/...
 * /velodyne_nodelet_manager_cloud/max_range: 130.0
 * /velodyne_nodelet_manager_cloud/min_range: 0.4
 * /velodyne_nodelet_manager_driver/device_ip: 
 * /velodyne_nodelet_manager_driver/frame_id: velodyne
 * /velodyne_nodelet_manager_driver/model: VLP16
 * /velodyne_nodelet_manager_driver/pcap: 
 * /velodyne_nodelet_manager_driver/port: 2368
 * /velodyne_nodelet_manager_driver/read_fast: False
 * /velodyne_nodelet_manager_driver/read_once: False
 * /velodyne_nodelet_manager_driver/repeat_delay: 0.0
 * /velodyne_nodelet_manager_driver/rpm: 600.0
 * /velodyne_nodelet_manager_laserscan/resolution: 0.007
 * /velodyne_nodelet_manager_laserscan/ring: -1

NODES
  /
    velodyne_nodelet_manager (nodelet/nodelet)
    velodyne_nodelet_manager_cloud (nodelet/nodelet)
    velodyne_nodelet_manager_driver (nodelet/nodelet)
    velodyne_nodelet_manager_laserscan (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [10972]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to fa4bc318-b63d-11e8-a5fd-34e12d206854
process[rosout-1]: started with pid [10985]
started core service [/rosout]
process[velodyne_nodelet_manager-2]: started with pid [10999]
process[velodyne_nodelet_manager_driver-3]: started with pid [11003]
process[velodyne_nodelet_manager_cloud-4]: started with pid [11004]
process[velodyne_nodelet_manager_laserscan-5]: started with pid [11005]
[FATAL] [1536723796.245717709]: Failed to load nodelet '/velodyne_nodelet_manager_cloud` of type `velodyne_pointcloud/CloudNodelet` to manager `velodyne_nodelet_manager'
[FATAL] [1536723796.245717289]: Failed to load nodelet '/velodyne_nodelet_manager_driver` of type `velodyne_driver/DriverNodelet` to manager `velodyne_nodelet_manager'
[FATAL] [1536723796.245733096]: Failed to load nodelet '/velodyne_nodelet_manager_laserscan` of type `velodyne_laserscan/LaserScanNodelet` to manager `velodyne_nodelet_manager'
[velodyne_nodelet_manager-2] process has died [pid 10999, exit code -11, cmd /opt/ros/kinetic/lib/nodelet/nodelet manager __name:=velodyne_nodelet_manager __log:=/home/laura/.ros/log/fa4bc318-b63d-11e8-a5fd-34e12d206854/velodyne_nodelet_manager-2.log].
log file: /home/laura/.ros/log/fa4bc318-b63d-11e8-a5fd-34e12d206854/velodyne_nodelet_manager-2*.log
[velodyne_nodelet_manager_driver-3] process has died [pid 11003, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load velodyne_driver/DriverNodelet velodyne_nodelet_manager __name:=velodyne_nodelet_manager_driver __log:=/home/laura/.ros/log/fa4bc318-b63d-11e8-a5fd-34e12d206854/velodyne_nodelet_manager_driver-3.log].
log file: /home/laura/.ros/log/fa4bc318-b63d-11e8-a5fd-34e12d206854/velodyne_nodelet_manager_driver-3*.log
[velodyne_nodelet_manager_cloud-4] process has died [pid 11004, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load velodyne_pointcloud/CloudNodelet velodyne_nodelet_manager __name:=velodyne_nodelet_manager_cloud __log:=/home/laura/.ros/log/fa4bc318-b63d-11e8-a5fd-34e12d206854/velodyne_nodelet_manager_cloud-4.log].
log file: /home/laura/.ros/log/fa4bc318-b63d-11e8-a5fd-34e12d206854/velodyne_nodelet_manager_cloud-4*.log
[velodyne_nodelet_manager_laserscan-5] process has died [pid 11005, exit code 255, cmd /opt/ros/kinetic/lib/nodelet/nodelet load velodyne_laserscan/LaserScanNodelet velodyne_nodelet_manager __name:=velodyne_nodelet_manager_laserscan __log:=/home/laura/.ros/log/fa4bc318-b63d-11e8-a5fd-34e12d206854/velodyne_nodelet_manager_laserscan-5.log].
log file: /home/laura/.ros/log/fa4bc318-b63d-11e8-a5fd-34e12d206854/velodyne_nodelet_manager_laserscan-5*.log

If I clone the code, build locally, and source, it launches fine.
I have the following binary versions installed:

> ii  ros-kinetic-velodyne                         1.3.0-0xenial-20180824-165012-0800                    amd64        Basic ROS support for the Velodyne 3D LIDARs.
> ii  ros-kinetic-velodyne-driver                  1.3.0-0xenial-20180824-154954-0800                    amd64        ROS device driver for Velodyne 3D LIDARs.
> ii  ros-kinetic-velodyne-laserscan               1.3.0-0xenial-20180824-115138-0800                    amd64        Extract a single ring of a Velodyne PointCloud2 and publish it as a LaserScan message
> ii  ros-kinetic-velodyne-msgs                    1.3.0-0xenial-20180824-101236-0800                    amd64        ROS message definitions for Velodyne 3D LIDARs.
> ii  ros-kinetic-velodyne-pointcloud              1.3.0-0xenial-20180824-161402-0800                    amd64        Point cloud conversions for Velodyne 3D LIDARs.
`

lstelzner, 2018-09-13:
It looks to have been related to a recent kinetic release, and the fact all my other ros packages were an older version.  After upgrading to the new ROS packages, the velodyne launch started functioning as expected.  Closing.

closed 2018-09-13 by lstelzner

JWhitleyAStuff, 2018-09-13:
Sorry you had trouble but thanks for updating the issue! Glad you got it resolved!