ros-drivers/velodyne #188: Proper handling of missing NMEA string needed

moooeeeep, 2018-09-20:
When no NMEA sentences are received, the message handler throws exceptions. (I think this happens when it is attempted to create a `ros::Time(-1)` [here](https://github.com/ros-drivers/velodyne/blob/gps_imu/velodyne_gps_imu/src/gpsimu_driver.cc#L220).)

This prevents the IMU data to be published. I would suggest to add proper error handling there. Not sure what may be the most elegant approach to that.

Maybe just wrap the entire processing and publishing of NMEA stuff starting [here](https://github.com/ros-drivers/velodyne/blob/gps_imu/velodyne_gps_imu/src/gpsimu_driver.cc#L199) in something like this:

    if (nmeaHourUnixTime != -1.0) {
        // publish TimeReference
        // publish NMEA sentence
    }
    // publish IMU
    // publish Temperature

Or just wrap the relevant sections inside exception handlers ...

mrnabati, 2019-04-22:
The nmeaHourUnixTime function [here](https://github.com/ros-drivers/velodyne/blob/6883807f1807bef0c76a83c36a352f71a8806fef/velodyne_gps_imu/src/gpsimu_driver.cc#L199-L206) returns -1 if the sentence is incomplete, so the unsigned integer wraps around to a large number. By changing the datatype from a double to uint32, we can check if -1 is returned and return from the handlePacket function without publishing anything. 
```c++
uint32_t nmeaHourUnixTime = nmeaHourToUnixTime(nmea_sentence);
  if(nmeaHourUnixTime == -1){
      return false;
  }

  if (hour_time_ == 0 || vpp.gps_timestamp > 2.0) {
      hour_time_ = (double)nmeaHourUnixTime;
  }
```

JWhitleyAStuff, 2019-04-23:
@mrnabati - Would you mind submitting a PR with this change?

moooeeeep, 2019-04-23:
> The nmeaHourUnixTime function [here](https://github.com/ros-drivers/velodyne/blob/6883807f1807bef0c76a83c36a352f71a8806fef/velodyne_gps_imu/src/gpsimu_driver.cc#L199-L206) returns -1 if the sentence [...]

If done like this, the IMU message is also not published (that happens later in the body), when the NMEA string is invalid. Maybe adding some functions to split functionality properly would be in order.

Also comparing `uint32_t` to `int` without a cast, will trigger an unnecessary compiler warning.

JWhitleyAStuff, 2019-04-23:
@moooeeeep and @mrnabati - The `master` branch now contains the parameter `gps_time` (see https://github.com/ros-drivers/velodyne/blob/87c33ac81dae12b2eebe0007df615c91e196ca7b/velodyne_driver/src/lib/input.cc#L234 for implementation). This parameter allows you to timestamp the incoming packets based on GPS time from the sensor instead of an estimated ROS time based on the scan start. I think it might be a good time to migrate some of the features in the `gps_imu` branch into `velodyne_driver` instead. Since we receive the GPS data and process it in `master` already, would one of you mind making a PR that just adds a publisher for sensor_msgs::TimeReference to `velodyne_driver` in `master`? I think adding these publishers to `velodyne_driver` instead of a separate package makes more sense since, IMO, a "driver" should interpret all raw data coming from the sensor, not just the primarily-used data (lidar points).

JWhitleyWork, 2020-04-29:
Closing this issue since the `gps_time` parameter was implemented in another PR. If there is any further discussion, feel free to re-open.

closed 2020-04-29 by JWhitleyWork