ros-drivers/velodyne #190: VLP-16 vert_offset_correction all zero in YAML

msemtd, 2018-10-08:
The VLP-16 User Manual Table 9-1 shows various vertical offset corrections but these don't appear in        velodyne/velodyne_pointcloud/params/VLP16db.yaml

NB: I know that this is generated by velodyne/velodyne_pointcloud/scripts/gen_calibration.py and that the VLP-16 XML calibration file for Veloview also has zero vertical offset corrections (see VeloView/share/VLP-16.xml) - I just want to know what is correct!

kaarta-SHanna, 2019-12-09:
I also see this as being a potential issue. For reference, here is the section of the manual that explains this:

[image]
[image]

jerry73204, 2020-01-13:
Interestingly, the manual does not mention the actual calculation of the point positions using vertical corrections, nor explain the meaning of vertical correction. The only reference I can see is [rawdata.cc](https://github.com/ros-drivers/velodyne/blob/2e4e58259ddcf5c40e678eaac957b12114a06f07/velodyne_pointcloud/src/lib/rawdata.cc#L328). The state is even in limbo even that the yaml files does not respect the manual.

YoshuaNava, 2021-02-21:
Hi,
I opened an issue with the same question (https://github.com/ros-drivers/velodyne/issues/385).

In this regard, we could do two things: 1) Ask Velodyne support directly, 2) Test the impact of these parameters, by adding them to the point coordinates computation.

maciej-3, 2021-04-06:
> Hi,
> I opened an issue with the same question (#385).
> 
> In my view, we could: 1) Ask Velodyne support directly, 2) Check if other Velodyne driver implementations (e.g. Veloview, KumarRobotics/VelodynePuck) employ them, 3) Test the impact of these parameters, by adding them to the point coordinates computation.
> 
> I just submitted a support request to Velodyne Lidar. Can someone test their impact on the data?

Hi! Is there any update on this?

maciej-3, 2021-04-07:
Ok, I have just investigated this issue briefly and I think that both Matlab drivers and ROS drivers implement the vertical correction in the same way.

Why?
I scanned a planar target of a diamond shape and filtered the point cloud both in Matlab and in ROS so that I get only the last ring (laser ID = 15 according to the [manual](https://greenvalleyintl.com/wp-content/uploads/2019/02/Velodyne-LiDAR-VLP-16-User-Manual.pdf)). Then I computed the mean of the z coordinates of the pointclouds and they are the same up to 1mm. This means that if the vertical correction was wrongly implemented in ROS, the mean would differ by around 11mm.

This brief investigation does not obviously mean that both Matlab and ROS drivers are implemented correctly, but it means that drivers yield the same values on the Z-axis. This, in turn, increases the probability that they are correct given that the guys from Matlab would have spotted this issue when they implemented their driver since they are paid for this job.

YoshuaNava, 2021-04-07:
Hey,

Yes. My support request received the following reply:

```
Hello Yoshua,

I talked with a member on my team about this and the reason these values are not present in the xml file is because they are negligible since our sensors is accurate to +/- 3cm. 

Best Regards,
```

The above means that the calibration values from the original source do neglect those vertical offsets.

closed 2022-12-04 by JWhitleyWork