ros-drivers/velodyne #191: VLP16 timestamp not working

egoist-sx, 2018-10-15:
I have been working with Velodyne VLP16 recently and trying to get timestamp of each data packet.

According to VLP16's documentation, there are four byte timestamp represent microsecond since top of hour.
Currently in rawdata.h: raw_packet struct contains uint16_t revolution. But suppose it be uint32_t timestamp? 

Otherwise raw_packet's size does not match UDP packet content size. I have searched through VLP16's documentation, it did not mention anything about revolution.

msemtd, 2018-10-15:
The uint32_t timestamp can be found in each packet at byte index 1200. See figure 9-2 in the VLP-16 User Manual. The manual does not mention revolution being part of the packet data and the rawdata.h heading is not clear which device manual it is talking about - it's clearly wrong for the VLP-16!

A better version might be something like 
`  typedef struct raw_packet
  {
    raw_block_t blocks[BLOCKS_PER_PACKET];
    uint32_t timestamp;
    uint8_t factory1;
    uint8_t factory2;
  } raw_packet_t;
`

egoist-sx, 2018-10-16:
@msemtd Thanks a lot! I have modified the code and checked timestamp read from VLP-16 but it differs a lot from ROS time on how many seconds past since last hour. Do you observe something similar? Currently I am using ROS time stamp by adding offset between those two measured at first received packet.

egoist-sx, 2018-10-16:
@JWhitleyAStuff Thank you very much.

closed 2018-10-16 by egoist-sx