ros-drivers/velodyne #192: velodyne_laserscan only works for the one ring close to the horizontal

jviramo1, 2018-10-24:
When using the velodyne_laserscan node we're expecting to see the resulting scan line up with the points in the pointcloud. However, with VLP-16 this is only true whenever ring 8 is selected with the errors being more evident with the rings at the two extremes (ring 0 and 15 for VLP-16). 
Looking at the driver it appears to me that the conversion does not account for the fact that other rings are not aligned with the frame of the device.
If this is by design, then a note should be added to avoid confusions. We were lucky that the error was very noticeable with the ring we were using but this might not be the case with rings that are close to the mid ring.

JWhitleyAStuff, 2018-11-15:
@jviramo1 - I believe the idea was that each individual ring would assume an origin at the central point of it's conical projection. I would need clarification from @kmhallen to be sure. I've assigned him this issue so we should hopefully hear back soon.

kmhallen, 2018-11-15:
This is by design. The [LaserScan](http://docs.ros.org/api/sensor_msgs/html/msg/LaserScan.html) message doesn't have the 3D concept of height. For this reason, the 3D point is projected to the x-y plane by assuming z is zero. It is only intended to be used with the ring closest to zero elevation. The conversion from a 3D PointCloud to a 2D LaserScan will inherently lose information.

I should probably document the velodyne_laserscan package on the wiki page.

jviramo1, 2018-11-16:
Thanks for the clarification. 
My impression was that the node would project the points on an x-y plane aligned with the elevation angle of each individual ring, then you wouldn't lose any information going into a 2D LaserScan. In other words, we would assume that a VLP-16 is a collection of 16 lasers each with a different elevation angle. 
In addition to documenting the velodyne_laserscan package I think it would be helpful to remove the code that allows the user to select a ring since code is only valid for the default ring (i.e. the one closest to the 0 elevation angle).

closed 2018-11-16 by jviramo1

kmhallen, 2018-11-16:
There is no way to do that. The rings form cones like the top diagram, not planes like the bottom diagram. The LaserScan message does not have an elevation angle value, so it must be a plane.

I think the ring selection code should stay so this can be used with other LIDAR sensors where the horizontal ring number is not known at this time.
[image]

JWhitleyAStuff, 2018-11-16:
@kmhallen There are two other options: 

1. You could make the selected ring dependent upon the type of sensor selected. Since there are only a fixed number of Velodyne sensors and each "perpendicular ring" is known for each sensor, this would cut down significantly on confusion, IMO. 
2. You could "flatten" each conical projection about a ring on the cone defined by the distance from the sensor's origin to the physical laser. This would necessarily cause a warping of the world-space defined by the points but it would create a perfectly horizontal line of points. Not sure what use this has.

kmhallen, 2018-11-16:
1. We could calculate the average angle of elevation of each ring for the first message, and use the one closest to zero as the default. This would support all sensors without prior knowledge. However, most sensors do not have a horizontal ring, and the two that are above and below horizontal are equally valid.

1. We could optionally add z into the range measurement, but only if we can think of plausible use case. I can't think of one. It would not be a physically accurate representation.