ros-drivers/velodyne #193: RViz pointcloud flashing

lesterlo, 2018-10-31:
Hello, I am using this driver to visualize the point cloud on RViz, but the pointcloud on the RViz is flashing. I can only fix it by changing the rpm value in the launch file from 600 to 300, or the set the RViz visualize decay time from 0 to 0.15. However, the pointcloud will then become delayed and not shown in real time. 

I would greatly appreciate if there is any suggestion towards fixing the problem.

Lidar Model: HDL-64E S3
Rpm: 600 (default)

launch file setting
Rpm: 600
Model parameter: 64E_S2
ROS: kinetic

lesterlo, 2018-10-31:
Ok. I solve the problem. Seems my HDL-64E_S3 is in Dual return mode. I add a new config model called 64E-S3 on [this line](https://github.com/ros-drivers/velodyne/blob/797eead188f16988f738dd4efef394d94b55b683/velodyne_driver/src/driver/driver.cc#L40) . I assume the lidar generates 2233312 points per second in dual fire mode.  I change the packet rate from 3472.17 to 5815.91 (2233312/384). And the flashing problem is gone.

JWhitleyAStuff, 2018-12-05:
Created issue to support dual-return mode in the future. Thanks.

closed 2018-12-05 by JWhitleyAStuff