ros-drivers/velodyne #198: Testing and Support for Dual-Return Mode

JWhitleyAStuff, 2018-12-05:
Dual-return mode does not seem to be properly implemented (as mentioned by #109 and shown by the problems enabling it causes in #193). This is a reminder item to eventually add full, working support for dual-return mode to the driver.

andre-nguyen, 2018-12-20:
@JWhitleyAStuff I'd like to look into this, but it is unclear to me how this would be implemented such that it works correctly for all lidar models. Any pointers?

JWhitleyAStuff, 2018-12-20:
This isn't by any means the only way but, if I was handling this, I would do the following:

- Create a launch file parameter called "dual_return_mode" with true/false values.
- Modify the `packet_rate` values for each sensor in `velodyne_driver` in driver.cc if the above launch parameter is true. For the HDL-64E S3, see page 50 of this document: https://velodynelidar.com/lidar/products/manual/HDL-64E%20S3%20manual.pdf For the HDL-64E S2, see this document: https://www.velodynelidar.com/lidar/products/manual/63-HDL64E%20S2%20Manual_Rev%20D_2011_web.pdf For the VLP-32C, see page 50 of this document: https://velodynelidar.com/docs/manuals/63-9325%20REV%20B%20MANUAL,USERS,VLP-32.pdf For the HDL-32E, see page 54 of this document: https://velodynelidar.com/docs/manuals/63-9113%20REV%20M%20MANUAL,USERS,HDL32E.pdf For the VLP-16, see page 50 of this document: https://velodynelidar.com/docs/manuals/63-9243%20REV%20D%20MANUAL,USERS,VLP-16.pdf
- The function `RawData::unpack` in rawdata.h and rawdata.cc in `velodyne_pointcloud` is where the data are unpacked from the driver-published scans. Unfortunately, some of the values here are set as constants (like the size of SCANS_PER_BLOCK which is used in calculating the BLOCK_DATA_SIZE which is used in the raw_packet_t struct) which will need to be changed to variables and doubled if the parameter is true.
- Create two point clouds if the above parameter is true.
- Either publish the usual `velodyne_points` if the above launch parameter is false or publish `velodyne_points_last` and `velodyne_points_strongest` if it is true.

Here are some documents that should help with understanding the modifications to the data structures in dual return mode:

https://velodynelidar.com/docs/notes/63-9277%20Rev%20B%20HDL-32E%20Application%20Note%20-%20Packet%20Structure%20&%20Timing%20Definition.pdf (see the page "Data Block Structure in Dual Return Mode")

https://velodynelidar.com/docs/manuals/63-HDL64ES3%20REV%20J%20MANUAL,USERS%20AND%20PROGRAM%20GUIDE,HDL-64E%20S3.pdf (see Dual Returns on page 15)

https://icave2.cse.buffalo.edu/resources/sensor-modeling/VLP32CManual.pdf (see page 57)

I hope this helps. Please let me know if you have further questions.

andre-nguyen, 2018-12-20:
@JWhitleyAStuff Excellent pointers, I can definitely look into the VLP16 part as that's what I have on hand, we'll see later for the others.

JWhitleyAStuff, 2018-12-22:
1. The downside to this method is that you won't know until you receive the first couple of blocks which means the topics won't be available until after that. You could do this and publish all three with /points containing all of them and the other two just containing those types. Doesn't matter much to me. Either way is fine. 
2. I would publish the double points on both topics because, if someone is only using one of the topics as their points source, they will be missing those returns.

w4rlock999, 2020-06-30:
Is this resolved ? I just realized there is no option for dual-return in the parameter of the driver, sorry to bump an old thread

JWhitleyWork, 2020-07-28:
@w4rlock999 Unfortunately, no. There was a start on it but the developer was waiting on me to provide PCAP files from several sensors and I ended up leaving the job where I had access to them in order to test it. Would you be interested in working on the feature?

FSund, 2022-10-04:
Are the changes in #208 still relevant for supporting dual return, or has `master` changed too much? 

I have access to a VLP-16 at work, and we are thinking about experimenting with dual return.