ros-drivers/velodyne #201: Provide Documentation for velodyne_laserscan

JWhitleyAStuff, 2018-12-10:
@kmhallen - Would you mind providing some documentation on http://wiki.ros.org/velodyne_laserscan and adding a README.md to this package which points to that URL?

kmhallen, 2018-12-10:
Sure, I'll try to take care of it over the holidays

JWhitleyAStuff, 2018-12-10:
Excellent. Thank you!

JWhitleyAStuff, 2020-01-22:
@kmhallen - Any update? We got an external request for this as well. See #309.

mfkenson, 2020-05-10:
I want the doc. At the mean time I followed the launch file here (see the params)

https://github.com/ros-drivers/velodyne/blob/master/velodyne_pointcloud/launch/laserscan_nodelet.launch

JWhitleyWork, 2020-05-11:
@kmhallen Are you still able to do this? There have been a significant number of requests and I can't make time to create the docs.