ros-drivers/velodyne #202: Laser beam emission interval

pavankumarbn, 2018-12-12:
Hi All,

I am using  Velodyne Puck Lite for PCD acquisition, but after the scan, data appears to be distorted horizontally, Projected points are not clearly aligned with each channel data.  V
[image]
elodyne is mounted on Drone to acquire the data.

This is  the launch file

  `

<launch>
  
  <arg name="calibration" default="$(find velodyne_pointcloud)/params/VLP16db.yaml"/>
  <arg name="device_ip" default="" />
  <arg name="frame_id" default="velodyne" />
  <arg name="manager" default="$(arg frame_id)_nodelet_manager" />
  <arg name="max_range" default="130.0" />
  <arg name="min_range" default="0.4" />
  <arg name="pcap" default="" />
  <arg name="port" default="2368" />
  <arg name="read_fast" default="false" />
  <arg name="read_once" default="false" />
  <arg name="repeat_delay" default="0.0" />
  <arg name="rpm" default="300.0" />
  <arg name="cut_angle" default="-0.01" />
  <arg name="laserscan_ring" default="-1" />
  <arg name="laserscan_resolution" default="0.007" />

  
  <include file="$(find velodyne_driver)/launch/nodelet_manager.launch">
    <arg name="device_ip" value="$(arg device_ip)"/>
    <arg name="frame_id" value="$(arg frame_id)"/>
    <arg name="manager" value="$(arg manager)" />
    <arg name="model" value="VLP16"/>
    <arg name="pcap" value="$(arg pcap)"/>
    <arg name="port" value="$(arg port)"/>
    <arg name="read_fast" value="$(arg read_fast)"/>
    <arg name="read_once" value="$(arg read_once)"/>
    <arg name="repeat_delay" value="$(arg repeat_delay)"/>
    <arg name="rpm" value="$(arg rpm)"/>
    <arg name="cut_angle" value="$(arg cut_angle)"/>
  </include>

  
  <include file="$(find velodyne_pointcloud)/launch/cloud_nodelet.launch">
    <arg name="calibration" value="$(arg calibration)"/>
    <arg name="manager" value="$(arg manager)" />
    <arg name="max_range" value="$(arg max_range)"/>
    <arg name="min_range" value="$(arg min_range)"/>
  </include>

  
  <include file="$(find velodyne_pointcloud)/launch/laserscan_nodelet.launch">
    <arg name="manager" value="$(arg manager)" />
    <arg name="ring" value="$(arg laserscan_ring)"/>
    <arg name="resolution" value="$(arg laserscan_resolution)"/>
  </include>

</launch>`

////// calibrationfile   VLP16db.yaml

`lasers:
- {dist_correction: 0.0, dist_correction_x: 0.0, dist_correction_y: 0.0, focal_distance: 0.0,
  focal_slope: 0.0, horiz_offset_correction: 0.0, laser_id: 0, rot_correction: 0.0,
  vert_correction: -0.2617993877991494, vert_offset_correction: 0.0}
- {dist_correction: 0.0, dist_correction_x: 0.0, dist_correction_y: 0.0, focal_distance: 0.0,
  focal_slope: 0.0, horiz_offset_correction: 0.0, laser_id: 1, rot_correction: 0.0,
  vert_correction: 0.017453292519943295, vert_offset_correction: 0.0}
- {dist_correction: 0.0, dist_correction_x: 0.0, dist_correction_y: 0.0, focal_distance: 0.0,
  focal_slope: 0.0, horiz_offset_correction: 0.0, laser_id: 2, rot_correction: 0.0,
  vert_correction: -0.22689280275926285, vert_offset_correction: 0.0}
- {dist_correction: 0.0, dist_correction_x: 0.0, dist_correction_y: 0.0, focal_distance: 0.0,
  focal_slope: 0.0, horiz_offset_correction: 0.0, laser_id: 3, rot_correction: 0.0,
  vert_correction: 0.05235987755982989, vert_offset_correction: 0.0}
- {dist_correction: 0.0, dist_correction_x: 0.0, dist_correction_y: 0.0, focal_distance: 0.0,
  focal_slope: 0.0, horiz_offset_correction: 0.0, laser_id: 4, rot_correction: 0.0,
  vert_correction: -0.19198621771937624, vert_offset_correction: 0.0}
- {dist_correction: 0.0, dist_correction_x: 0.0, dist_correction_y: 0.0, focal_distance: 0.0,
  focal_slope: 0.0, horiz_offset_correction: 0.0, laser_id: 5, rot_correction: 0.0,
  vert_correction: 0.08726646259971647, vert_offset_correction: 0.0}
- {dist_correction: 0.0, dist_correction_x: 0.0, dist_correction_y: 0.0, focal_distance: 0.0,
  focal_slope: 0.0, horiz_offset_correction: 0.0, laser_id: 6, rot_correction: 0.0,
  vert_correction: -0.15707963267948966, vert_offset_correction: 0.0}
- {dist_correction: 0.0, dist_correction_x: 0.0, dist_correction_y: 0.0, focal_distance: 0.0,
  focal_slope: 0.0, horiz_offset_correction: 0.0, laser_id: 7, rot_correction: 0.0,
  vert_correction: 0.12217304763960307, vert_offset_correction: 0.0}
- {dist_correction: 0.0, dist_correction_x: 0.0, dist_correction_y: 0.0, focal_distance: 0.0,
  focal_slope: 0.0, horiz_offset_correction: 0.0, laser_id: 8, rot_correction: 0.0,
  vert_correction: -0.12217304763960307, vert_offset_correction: 0.0}
- {dist_correction: 0.0, dist_correction_x: 0.0, dist_correction_y: 0.0, focal_distance: 0.0,
  focal_slope: 0.0, horiz_offset_correction: 0.0, laser_id: 9, rot_correction: 0.0,
  vert_correction: 0.15707963267948966, vert_offset_correction: 0.0}
- {dist_correction: 0.0, dist_correction_x: 0.0, dist_correction_y: 0.0, focal_distance: 0.0,
  focal_slope: 0.0, horiz_offset_correction: 0.0, laser_id: 10, rot_correction: 0.0,
  vert_correction: -0.08726646259971647, vert_offset_correction: 0.0}
- {dist_correction: 0.0, dist_correction_x: 0.0, dist_correction_y: 0.0, focal_distance: 0.0,
  focal_slope: 0.0, horiz_offset_correction: 0.0, laser_id: 11, rot_correction: 0.0,
  vert_correction: 0.19198621771937624, vert_offset_correction: 0.0}
- {dist_correction: 0.0, dist_correction_x: 0.0, dist_correction_y: 0.0, focal_distance: 0.0,
  focal_slope: 0.0, horiz_offset_correction: 0.0, laser_id: 12, rot_correction: 0.0,
  vert_correction: -0.05235987755982989, vert_offset_correction: 0.0}
- {dist_correction: 0.0, dist_correction_x: 0.0, dist_correction_y: 0.0, focal_distance: 0.0,
  focal_slope: 0.0, horiz_offset_correction: 0.0, laser_id: 13, rot_correction: 0.0,
  vert_correction: 0.22689280275926285, vert_offset_correction: 0.0}
- {dist_correction: 0.0, dist_correction_x: 0.0, dist_correction_y: 0.0, focal_distance: 0.0,
  focal_slope: 0.0, horiz_offset_correction: 0.0, laser_id: 14, rot_correction: 0.0,
  vert_correction: -0.017453292519943295, vert_offset_correction: 0.0}
- {dist_correction: 0.0, dist_correction_x: 0.0, dist_correction_y: 0.0, focal_distance: 0.0,
  focal_slope: 0.0, horiz_offset_correction: 0.0, laser_id: 15, rot_correction: 0.0,
  vert_correction: 0.2617993877991494, vert_offset_correction: 0.0}
num_lasers: 16
distance_resolution: 0.002`

Anybody, please suggest fixing this issue. ? Thanks in Advance.

msemtd, 2018-12-12:
It may be worth applying the vert_offset_correction values mentioned in #190 

How does your scanner data look in VeloView?

pavankumarbn, 2018-12-13:
@msemtd Thank you for your reply. I will try to apply vert_offset_correction.

ajay1606, 2018-12-17:
@pavankumarbn , It seems to be a problem with the calibration file (.yaml) .Is that calibration file is default one?

pavankumarbn, 2018-12-17:
@ajay1606 Thanks for the reply. Yes, it is default one given with Velodyne Driver.

JWhitleyAStuff, 2018-12-17:
@ajay1606 - There should be an XML calibration file provided with every sensor from Velodyne. You can convert this XML file to a YAML file with the utility `velodyne_pointcloud/scripts/gen_calibration.py`.

pavankumarbn, 2018-12-18:
@msemtd @ajay1606 @JWhitleyAStuff Thank you all for your valuable suggestions. The reason for this problem is FOV restriction as mentioned by the Velodyne Tech Support Team that current Velodyne ROS drivers don't support FOV restriction.

JWhitleyAStuff, 2018-12-19:
@pavankumarbn - Please check out the cut_angle parameter on the driver. This is an in-software way to restrict the field of view. I'm not sure it will cover what you're looking for (it looks like it only starts at Y and goes around counter-clockwise) but it might help. I'm sorry the driver doesn't support this feature at this time but we might be able to do so in the ROS2 version or after organized cloud support is added.

closed 2018-12-19 by JWhitleyAStuff

pavankumarbn, 2018-12-19:
@JWhitleyAStuff Sure I will check! Thank you very much.

AaronWaxson, 2020-11-12:
Which driver can use with Puck Lite, I use velodyne_puck but it didn't work with Puck Lite. How should I do ?

nshahrestani, 2020-12-16:
@JWhitleyWork Is restricting the FOV using the velodyne GUI still now supported?

JWhitleyWork, 2020-12-16:
@nshahrestani Please do not bump old threads. If you have a question that has not been answered recently, please create a new issue or ask on [ROS Answers](https://answers.ros.org/).