ros-drivers/velodyne #209: Question

JangGyoungHoon, 2019-01-16:
Now, I have a multiple velodyne_packets topics in rosbag file. I want to re -logging rosbag with convert from velodyne_packets to velodyne_points. how to convert from velodyne_packets to velodyne_points in rosbag.

JangGyoungHoon, 2019-01-16:
I want to know how to run convert.cc (This code is conversion code from packets to points as my thought.)

JWhitleyAStuff, 2019-01-22:
@JangGyoungHoon - What you are trying to do is the purpose of the "cloud nodelet." The file velodyne_pointcloud/launch/cloud_nodelet.launch provides an example for launching this nodelet while passing in the name of a nodelet manager. This launch file is used by all of the sensor-specific launch files such as velodyne_pointcloud/launch/32e_points.launch.

closed 2019-01-22 by JWhitleyAStuff