ros-drivers/velodyne #211: error: undefined reference to `vtable for velodyne_pointcloud::PointcloudXYZIR'

Jie-Yuan-China, 2019-01-17:
hallo,
I have recorded the velodyne_msgs/VelodyneScan into rosbag, and i want to use it for accurate mapping task. now i want to transform it into pcl::pointcloud, but an error happens, any kind of help will be appreciated.
error info:
/home/***/catkin_ws/src/velodyne/velodyne_pointcloud/include/velodyne_pointcloud/pointcloudXYZIR.h:10: error: undefined reference to `vtable for velodyne_pointcloud::PointcloudXYZIR'
this is the code which i used for transformaing

` 

    void Scan_Analyser::convert_3d_lidar(
            const velodyne_msgs::VelodyneScan v_scan,
            const nav_msgs::Odometry gps_pose)
    {
        pcl::PointCloud<pcl::PointXYZI> current_cloud_3d;
        current_cloud_3d.clear();
        pcl::PointXYZI point;
        velodyne_rawdata::RawData data;
        data.setParameters(0.9,130.0,0.0,6.28);// TODO Change the parameters,
        data.setupOffline("file adresss",130.0,0.9);// TODO set calibration file, as well as max range, min range.
        velodyne_rawdata::VPointCloud::Ptr outMsg(new velodyne_rawdata::VPointCloud());
        outMsg->header.stamp = pcl_conversions::toPCL(v_scan.header).stamp;
    //    outMsg->header.frame_id = config_.frame_id;
    //    outMsg->height = 1;
        velodyne_pointcloud::PointcloudXYZIR v_point_cloud;
        for(size_t i=0;i<v_scan.packets.size();++i)
        {
            v_point_cloud.pc->points.clear();
            v_point_cloud.pc->width = 0;
            v_point_cloud.pc->height = 1;
            v_point_cloud.pc->header.stamp = pcl_conversions::toPCL(v_scan.header).stamp;
            v_point_cloud.pc->header.frame_id = v_scan.header.frame_id;
    
            //v_point_cloud.pc->points.reserve(v_scan->packets.size() * data.scansPerPacket());
            //pcl::PointCloud<velodyne_pointcloud::PointXYZIR> v_point_cloud;
            data.unpack(v_scan.packets[i],v_point_cloud);
            for(int j=0;j<v_point_cloud.pc->size();j++)
            {
                point.x=v_point_cloud.pc->points[j].x;
                point.y=v_point_cloud.pc->points[j].y;
                point.z=v_point_cloud.pc->points[j].z;
                point.intensity=v_point_cloud.pc->points[j].intensity;
                current_cloud_3d.push_back(point);
                std::cout<<"transform to PCL done"<<std::endl;
            }
        }
    }
`

JWhitleyAStuff, 2019-01-22:
@Jie-Yuan-China - What you are trying to do is the purpose of the "cloud nodelet." The file `velodyne_pointcloud/launch/cloud_nodelet` provides an example for launching this nodelet while passing in the name of a nodelet manager. This launch file is used by all of the sensor-specific launch files such as `velodyne_pointcloud/launch/32e_points.launch`.

hzdsdhr, 2020-04-07:
@JWhitleyWork Hi, is there a solution for the problem? I want to transfer laserscan into pointcloud, so I have to use the api of velodyne_pointcloud. I got the same problem. When I try to velodyne_pointcloud::PointcloudXYZIR v_point_cloud and catkin_make, it would give the same error.

Jie-Yuan-China, 2020-05-14:
> @JWhitleyWork Hi, is there a solution for the problem? I want to transfer laserscan into pointcloud, so I have to use the api of velodyne_pointcloud. I got the same problem. When I try to velodyne_pointcloud::PointcloudXYZIR v_point_cloud and catkin_make, it would give the same error.

Hi, sorry for replying so late, This Problem is solved by modifying the source related to "YAML" file. 
My Velodyne config file was compatible with an older version of the driver, after updating to a newer version it craps because of the loading of YAML file. I already forget how I changed the source code. but it is definitely related with YAML loader.  I hope this could be a little helpful to you.

JWhitleyWork, 2020-05-14:
Thanks for the update, @Jie-Yuan-China !