ros-drivers/velodyne #213: DriverNodelet's Poll Timeout Handling

jimmyw404, 2019-02-01:
In the DriverNodelet::devicePoll() function if the VelodyneDriver::poll() call fails, the devicePoll thread will exit. This happens if there is a network disconnect, the puck's power recycles etc.

This was requiring me to restart the velodyne Driver nodelet whenever I had a timeout.

Is the intent that there is some other mechanism to restart the DriverNodelet when it is detected it exits? What is the recommended way to do this?

For now I've commented out the break; line. My DriverNodelet::devicePoll() function looks like this:

`
/** @brief Device poll thread main loop. */
void DriverNodelet::devicePoll()
{
  while(ros::ok())
    {
      // poll device until end of file
      running_ = dvr_->poll();
      if (!running_)
      {

      	ROS_ERROR("W: DriverNodelet:devicePoll poll failed, who cares?");
//        break;
      }
    }
  running_ = false;
}`

And now it persists through disconnects and works great, but I'm guessing this is the inappropriate way to do this.

JWhitleyAStuff, 2019-02-14:
@jimmyw404 - Check the branch `maint/poll_timeout_handling`. Let me know if it works and I'll merge it if it does.

jimmyw404, 2019-02-18:
@JWhitleyAStuff Thank you for that code change. It will work just fine.

closed 2019-02-18 by jimmyw404