ros-drivers/velodyne #217: Catkin_make error: can't build a non-homogeneous workspace with non-catkin packages

etanx, 2019-02-21:
I was trying to install ROS drivers to process .pcap files from my VLP-32c on my Raspberry Pi 3 (running Raspbian Stretch). ROS kinetic has previously been installed. After searching how on earth to work with ROS packages, I tried:
```
cd ~/ros_catkin_ws/src
git clone -b master https://github.com/ros-drivers/velodyne.git
cd ~/ros_catkin_ws
catkin_make
```
... and got the output:
```
Base path: /home/pi/ros_catkin_ws
Source space: /home/pi/ros_catkin_ws/src
Build space: /home/pi/ros_catkin_ws/build
Devel space: /home/pi/ros_catkin_ws/devel
Install space: /home/pi/ros_catkin_ws/install
####
#### Running command: "cmake /home/pi/ros_catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/pi/ros_catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/pi/ros_catkin_ws/install -G Unix Makefiles" in "/home/pi/ros_catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/pi/ros_catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/kinetic
-- This workspace overlays: /opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/pi/ros_catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.8
CMake Warning (dev) at catkin/CMakeLists.txt:5 (project):
  Policy CMP0048 is not set: project() command manages VERSION variables.
  Run "cmake --help-policy CMP0048" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The following variable(s) would be set to empty:

catkin_VERSION
This warning is for project developers.  Use -Wno-dev to suppress it.

-- BUILD_SHARED_LIBS is on
WARNING: Package 'robot_model' is deprecated (This metapackage will be removed in ROS M. Replace all dependencies on
      &quot;robot_model&quot; in your package.xml with dependencies on collada_parser,
      collada_urdf, joint_state_publisher, kdl_parser, resource-retriever, urdf,
      urdf_parser_plugin, and liburdfdom-tools instead.)
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 194 packages in topological order:
-- ~~  - genmsg
-- ~~  - gencpp
-- ~~  - geneus
-- ~~  - genlisp
-- ~~  - gennodejs
-- ~~  - genpy
-- ~~  - bond_core (metapackage)
-- ~~  - cmake_modules
-- ~~  - class_loader
-- ~~  - common_msgs (metapackage)
-- ~~  - common_tutorials (metapackage)
-- ~~  - cpp_common
-- ~~  - desktop (metapackage)
-- ~~  - diagnostics (metapackage)
-- ~~  - eigen_stl_containers
-- ~~  - executive_smach (metapackage)
-- ~~  - geometry (metapackage)
-- ~~  - geometry_tutorials (metapackage)
-- ~~  - gl_dependency
-- ~~  - media_export
-- ~~  - message_generation
-- ~~  - message_runtime
-- ~~  - mk
-- ~~  - nodelet_core (metapackage)
-- ~~  - octomap (plain cmake)
-- ~~  - opencv3 (plain cmake)
-- ~~  - orocos_kdl (plain cmake)
-- ~~  - python_orocos_kdl (plain cmake)
-- ~~  - qt_dotgraph
-- ~~  - qt_gui
-- ~~  - qt_gui_py_common
-- ~~  - qwt_dependency
-- ~~  - random_numbers
-- ~~  - robot (metapackage)
-- ~~  - robot_model (metapackage)
-- ~~  - ros (metapackage)
-- ~~  - ros_base (metapackage)
-- ~~  - ros_comm (metapackage)
-- ~~  - ros_core (metapackage)
-- ~~  - ros_tutorials (metapackage)
-- ~~  - rosbag_migration_rule
-- ~~  - rosbash
-- ~~  - rosboost_cfg
-- ~~  - rosbuild
-- ~~  - rosclean
-- ~~  - roscpp_core (metapackage)
-- ~~  - roscpp_traits
-- ~~  - roscreate
-- ~~  - rosgraph
-- ~~  - roslang
-- ~~  - roslint
-- ~~  - roslisp
-- ~~  - rosmake
-- ~~  - rosmaster
-- ~~  - rospack
-- ~~  - roslib
-- ~~  - rosparam
-- ~~  - rospy
-- ~~  - rosservice
-- ~~  - rostime
-- ~~  - roscpp_serialization
-- ~~  - python_qt_binding
-- ~~  - roslaunch
-- ~~  - rosunit
-- ~~  - angles
-- ~~  - rosconsole
-- ~~  - pluginlib
-- ~~  - qt_gui_cpp
-- ~~  - resource_retriever
-- ~~  - rosconsole_bridge
-- ~~  - roslz4
-- ~~  - rosbag_storage
-- ~~  - rostest
-- ~~  - rqt_action
-- ~~  - rqt_bag
-- ~~  - rqt_bag_plugins
-- ~~  - rqt_common_plugins (metapackage)
-- ~~  - rqt_console
-- ~~  - rqt_dep
-- ~~  - rqt_graph
-- ~~  - rqt_gui
-- ~~  - rqt_logger_level
-- ~~  - rqt_moveit
-- ~~  - rqt_msg
-- ~~  - rqt_nav_view
-- ~~  - rqt_plot
-- ~~  - rqt_pose_view
-- ~~  - rqt_publisher
-- ~~  - rqt_py_console
-- ~~  - rqt_reconfigure
-- ~~  - rqt_robot_dashboard
-- ~~  - rqt_robot_monitor
-- ~~  - rqt_robot_plugins (metapackage)
-- ~~  - rqt_robot_steering
-- ~~  - rqt_runtime_monitor
-- ~~  - rqt_service_caller
-- ~~  - rqt_shell
-- ~~  - rqt_srv
-- ~~  - rqt_tf_tree
-- ~~  - rqt_top
-- ~~  - rqt_topic
-- ~~  - rqt_web
-- ~~  - smach
-- ~~  - smclib
-- ~~  - std_msgs
-- ~~  - actionlib_msgs
-- ~~  - bond
-- ~~  - diagnostic_msgs
-- ~~  - geometry_msgs
-- ~~  - eigen_conversions
-- ~~  - kdl_conversions
-- ~~  - nav_msgs
-- ~~  - rosgraph_msgs
-- ~~  - rosmsg
-- ~~  - rqt_py_common
-- ~~  - sensor_msgs
-- ~~  - map_msgs
-- ~~  - shape_msgs
-- ~~  - smach_msgs
-- ~~  - std_srvs
-- ~~  - stereo_msgs
-- ~~  - tf2_msgs
-- ~~  - tf2
-- ~~  - trajectory_msgs
-- ~~  - control_msgs
-- ~~  - urdf_parser_plugin
-- ~~  - velodyne (metapackage)
-- ~~  - velodyne_msgs
-- ~~  - visualization_msgs
-- ~~  - geometric_shapes
-- ~~  - visualization_tutorials (metapackage)
-- ~~  - viz (metapackage)
-- ~~  - webkit_dependency
-- ~~  - xmlrpcpp
-- ~~  - roscpp
-- ~~  - bondcpp
-- ~~  - bondpy
-- ~~  - nodelet
-- ~~  - nodelet_tutorial_math
-- ~~  - pluginlib_tutorials
-- ~~  - roscpp_tutorials
-- ~~  - rosout
-- ~~  - cv_bridge
-- ~~  - diagnostic_aggregator
-- ~~  - diagnostic_updater
-- ~~  - diagnostic_common_diagnostics
-- ~~  - dynamic_reconfigure
-- ~~  - filters
-- ~~  - joint_state_publisher
-- ~~  - message_filters
-- ~~  - image_transport
-- ~~  - rosnode
-- ~~  - rospy_tutorials
-- ~~  - rostopic
-- ~~  - roswtf
-- ~~  - rqt_gui_cpp
-- ~~  - rqt_gui_py
-- ~~  - rqt_image_view
-- ~~  - self_test
-- ~~  - smach_ros
-- ~~  - tf2_py
-- ~~  - topic_tools
-- ~~  - rosbag
-- ~~  - actionlib
-- ~~  - actionlib_tutorials
-- ~~  - diagnostic_analysis
-- ~~  - nodelet_topic_tools
-- ~~  - rqt_launch
-- ~~  - tf2_ros
-- ~~  - tf
-- ~~  - interactive_markers
-- ~~  - interactive_marker_tutorials
-- ~~  - laser_geometry
-- ~~  - tf2_kdl
-- ~~  - tf_conversions
-- ~~  - turtlesim
-- ~~  - turtle_actionlib
-- ~~  - turtle_tf
-- ~~  - turtle_tf2
-- ~~  - urdf
-- ~~  - collada_parser
-- ~~  - collada_urdf
-- ~~  - kdl_parser
-- ~~  - robot_state_publisher
-- ~~  - rviz
-- ~~  - librviz_tutorial
-- ~~  - rqt_rviz
-- ~~  - rviz_plugin_tutorials
-- ~~  - rviz_python_tutorial
-- ~~  - velodyne_driver
-- ~~  - velodyne_laserscan
-- ~~  - velodyne_pointcloud
-- ~~  - visualization_marker_tutorials
-- ~~  - xacro
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
CMake Error at catkin/cmake/catkin_workspace.cmake:95 (message):
  This workspace contains non-catkin packages in it, and catkin cannot build
  a non-homogeneous workspace without isolation.  Try the
  'catkin_make_isolated' command instead.
Call Stack (most recent call first):
  CMakeLists.txt:63 (catkin_workspace)

-- Configuring incomplete, errors occurred!
See also "/home/pi/ros_catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/pi/ros_catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
```
Some other forums mentioned to delete existing folders `build`, `devel`, and `install` in the catkin folder, but that didn't solve the problem. Trying `catkin_make_isolated` as the output suggested didn't work either hmm.
```
Base path: /home/pi/ros_catkin_ws
Source space: /home/pi/ros_catkin_ws/src
Build space: /home/pi/ros_catkin_ws/build_isolated
Traceback (most recent call last):
  File "/opt/ros/kinetic/bin/catkin_make_isolated", line 160, in <module>
    main()
  File "/opt/ros/kinetic/bin/catkin_make_isolated", line 156, in main
    override_build_tool_check=opts.override_build_tool_check,
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/catkin/builder.py", line 831, in build_workspace_isolated
    mark_space_as_built_by(buildspace, 'catkin_make_isolated')
  File "/usr/lib/python2.7/dist-packages/catkin_pkg/tool_detection.py", line 78, in mark_space_as_built_by
    with open(marker_path, 'w') as f:
IOError: [Errno 13] Permission denied: '/home/pi/ros_catkin_ws/build_isolated/.built_by'
```
Any suggestions on getting the GitHub ROS drivers to install, or perhaps another way to install them?

JWhitleyAStuff, 2019-02-21:
There are several issues in your workspace but none of them are related to the Velodyne driver. The first issue I see is that catkin says you're building packages in your workspace which are not catkin packages. The build system prior to catkin was rosbuild and catkin does not properly support building rosbuild packages. You can tell them apart because catkin uses package.xml files and rosbuild used Manifest files.  Secondly, the last line in your output indicates a file permissions issue. Make sure you aren't using sudo anywhere in your build process. Lastly, you are going to get INCREDIBLY poor performance on a Raspberry Pi trying to process files as dense as those generated by a VLP-32C. I highly recommend some different hardware. Good luck.

closed 2019-02-21 by JWhitleyAStuff

amadojoao, 2019-10-01:
I have the same problem, please reopen the topic

francisco-mercado, 2020-10-06:
I solved the issue by doing in the workspace directory:

$ which catkin_make_isolated
# this printed out for me : /home/pi/ros_catkin_ws/devel/bin/catkin_make_isolated
# and then
$ sudo /home/pi/ros_catkin_ws/devel/bin/catkin_make_isolated

JWhitleyWork, 2020-10-06:
@amadojoao Could you please describe your situation in more detail or open a new issue?

raypsaw, 2021-08-15:
I've the same issue that I try

  cd <ws>/src
  git clone https://github.com/ros-drivers/rosserial.git
  cd <ws>
  catkin_make
  catkin_make install

and got the output:

Base path: /home/robotis/catkin_ws
Source space: /home/robotis/catkin_ws/src
Build space: /home/robotis/catkin_ws/build
Devel space: /home/robotis/catkin_ws/devel
Install space: /home/robotis/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/robotis/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/robotis/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/robotis/catkin_ws/devel;/opt/ros/kinetic
-- This workspace overlays: /home/robotis/catkin_ws/devel;/opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/robotis/catkin_ws/build/test_results
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.14
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 69 packages in topological order:
-- ~~  - humanoid_navigation (metapackage)
-- ~~  - op3_bringup
-- ~~  - op3_description
-- ~~  - op3_gazebo
-- ~~  - op3_navigation
-- ~~  - op3_web_setting_tool
-- ~~  - robotis_framework (metapackage)
-- ~~  - robotis_op3 (metapackage)
-- ~~  - robotis_op3_common (metapackage)
-- ~~  - robotis_op3_demo (metapackage)
-- ~~  - robotis_op3_msgs (metapackage)
-- ~~  - robotis_op3_tools (metapackage)
-- ~~  - robotis_utility (metapackage)
-- ~~  - rosserial (metapackage)
WARNING: The CMakeLists.txt of the metapackage 'rosserial' contains non standard content. Use the content of the following file instead: /home/robotis/catkin_ws/build/catkin_generated/metapackages/rosserial/CMakeLists.txt
-- ~~  - rosserial_arduino
-- ~~  - rosserial_chibios
-- ~~  - rosserial_mbed
-- ~~  - rosserial_msgs
-- ~~  - rosserial_python
-- ~~  - rosserial_tivac
-- ~~  - rosserial_vex_cortex
-- ~~  - rosserial_vex_v5
-- ~~  - rosserial_xbee
-- ~~  - rosserial_client
-- ~~  - op3_action_module_msgs
-- ~~  - op3_offset_tuner_msgs
-- ~~  - op3_tuning_module_msgs
-- ~~  - op3_walking_module_msgs
-- ~~  - beginner_tutorials
-- ~~  - dynamixel_sdk
-- ~~  - gridmap_2d
-- ~~  - humanoid_planner_2d
-- ~~  - robotis_device
-- ~~  - robotis_framework_common
-- ~~  - robotis_math
-- ~~  - op3_balance_control
-- ~~  - op3_kinematics_dynamics
-- ~~  - ros_madplay_player
-- ~~  - ros_mpg321_player
-- ~~  - op3_offset_tuner_client
-- ~~  - op3_tuner_client
-- ~~  - scilab_optimization
-- ~~  - op3_camera_setting_tool
-- ~~  - rosserial_server
-- ~~  - face_detection
-- ~~  - op3_ball_detector
-- ~~  - op3_online_walking_module_msgs
-- ~~  - robotis_controller_msgs
-- ~~  - op3_action_module
-- ~~  - op3_base_module
-- ~~  - op3_direct_control_module
-- ~~  - op3_head_control_module
-- ~~  - op3_online_walking_module
-- ~~  - op3_read_write_demo
-- ~~  - op3_tuning_module
-- ~~  - op3_walking_module
-- ~~  - open_cr_module
-- ~~  - robotis_controller
-- ~~  - op3_action_editor
-- ~~  - op3_manager
-- ~~  - op3_offset_tuner_server
-- ~~  - footstep_planner
-- ~~  - op3_gui_demo
-- ~~  - op3_localization
-- ~~  - humanoid_localization
-- ~~  - rosserial_embeddedlinux
-- ~~  - rosserial_test
-- ~~  - rosserial_windows
-- ~~  - op3_demo
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkin_workspace.cmake:95 (message):
  This workspace contains non-catkin packages in it, and catkin cannot build
  a non-homogeneous workspace without isolation.  Try the
  'catkin_make_isolated' command instead.
Call Stack (most recent call first):
  CMakeLists.txt:63 (catkin_workspace)

-- Configuring incomplete, errors occurred!
See also "/home/robotis/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/robotis/catkin_ws/build/CMakeFiles/CMakeError.log".
Makefile:4434: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

please help me