ros-drivers/velodyne #218: VLP16 incorrect timestamp

ogail, 2019-02-27:
Hi,
We had a problem with VLP16 providing incorrect timestamps. Looking further in `rawdata.cc`, in `unpack` method we do following
```
const unsigned int* timestamp_since_last_hour = reinterpret_cast<const unsigned int*>(&pkt.data[1200]);
double packet_timestamp = static_cast<double>(*timestamp_since_last_hour) * 1e-6;
```

My expectation that the `packet_timestamp` should be always between `0` and `3600` seconds inclusive (as the timestamp is from last hour based on VLP-16 User Manual 63-9243 Rev. D). However I get this series of data for `packet_timestamp`
```
3599.98
3599.99
3600
48
3600.01
3600.02
3600.03
3600.05
3600.06
```
I wonder how come the `48` number came from and how come the number is also greater than 3600

JWhitleyAStuff, 2019-04-02:
I have not seen this issue and I believe you are correct about the timestamp_since_last_hour's expected values. Could you provide a more comprehensive log using the latest version of the driver? A PR was merged that allows using the actual packet time from the sensor to pass through instead of using an interpolated ROS time - please enable this feature with the `gps_time` parameter and see if the values change. I suspect that this is related to the time interpolation.

JWhitleyAStuff, 2019-05-15:
@ogail - Ping.

ogail, 2019-05-15:
We didn’t get a chance yet to update the driver, once happens will let you know

closed 2019-06-14 by JWhitleyAStuff

nisailhan, 2021-08-12:
> We didn’t get a chance yet to update the driver, once happens will let you know

Is there any update?