ros-drivers/velodyne #219: The startOrientation of the point cloud from rosbag is not consistent with pcap file

wwtinwhu, 2019-03-17:
Hi,Whitley. First of all, thank you for your contribution on velodyne-drivers. It is really good. But I have found a problem when I want to **calculate the start orientation of the pointcloud**.  

As usual, I use the function pcl::fromROSMsg() to get pointcloud. And I suppose the first element of the pointcloud is the **first point from the pointcloud** which is first fired and received because there is no more infomation. So next I use atan2() to calculate the orientation. I found that the value **varies a lot** between different frames. 

  So,next, I use VeloView to display the pacp file. I found that the start orientation of the frame is almost **around 0 degree**. 

  So,What do you think about this? How can I get the start orientation of the pointcloud? May be I can just use the 0 degree? Thank you very much! 

Picuture
This picture demonstrates that the start point of the frame is around 0 degree which is labeled red in the left. 
[image]

JWhitleyAStuff, 2019-03-17:
There are two reasons why the point clouds don't match the PCAP results: 
1. The point cloud is not organized - points are semi-random in the point cloud. 
2. A point could does not directly match one scan. The driver gathers points for a specified amount of time and publishes the cloud. It does not wait for the beginning or end of a scan.

@spuentz is working on number one in #214 but the other issue has not yet been addressed.

wwtinwhu, 2019-03-18:
> There are two reasons why the point clouds don't match the PCAP results:
> 
> 1. The point cloud is not organized - points are semi-random in the point cloud.
> 2. A point could does not directly match one scan. The driver gathers points for a specified amount of time and publishes the cloud. It does not wait for the beginning or end of a scan.
> 
> @spuetz is working on number one in #214 but the other issue has not yet been addressed.

Thanks,Whitley! There are two questions I have to ask. 
       

1.what does semi-random mean? The points in the pointcloud array are not arranged by time?

2.About the second reason you mentioned, I got that. You meant that pcap always waits for the beginning of a scan, and the driver don't. That can explain why the startOrientation of the point cloud is not around 0 degree(the start of pacp file). But there is a problem. **Why different frame received by driver in a continuous time have a different startOrientation?** The reason I supposed that is the semi-random you mention? 

Thank you very much!

JWhitleyAStuff, 2019-03-18:
@wwtinwhu:

1. Because the point clouds do not match up with the scans, the first point in the cloud will *likely* always be the same point within the scan but because the timing is not 100% identical to the scan timing, it may change. Organized point clouds will fix this problem. It will not fix the problem that the timing does not exactly match. This is why the startOrientation can be different and may change in the point clouds over time - we are using operating system timing which may be +/- a few milliseconds.

closed 2019-03-18 by JWhitleyAStuff

wwtinwhu, 2019-03-19:
@JWhitleyAStuff  That's so nice of you! I get that point, thank you to  solve my problem! Good luck!

zorosmith, 2019-06-13:
Hello @JWhitleyAStuff !
Can you obtain the organized point cloud in ros PointCloud format now?
I also meet this problem, plz share your method!
Any help will be appreciated!

JWhitleyAStuff, 2019-06-14:
@zorosmith - See my response in #248.

zorosmith, 2019-07-22:
> There are two reasons why the point clouds don't match the PCAP results:
>
> 1. The point cloud is not organized - points are semi-random in the point cloud.
> 2. A point could does not directly match [...]

Hello @JWhitleyAStuff , did you have any idea to obtain the same first point using this driver now?  I also meet with this problem. Look forward to your reply.

zorosmith, 2019-07-22:
Hello @wwtinwhu , How did you solve this problem to obtain the same first point finally? Could you share your method? Look forward to your reply.

JWhitleyAStuff, 2019-07-22:
@wwtinwhu / @zorosmith - Both of these issues have been resolved with new features. Setting the `organize_cloud` property to `true` and setting `cut_angle` to 6.28318530718 (2 × π) should make one point cloud equal one rotation and organize the cloud. In theory, this should make point index 0 always be the first point in each rotation. Let me know if this works for you!

zorosmith, 2019-07-22:
Thank you for your great work！ It works !

On 07/22/2019 19:01, Joshua Whitley wrote:

@wwtinwhu / @zorosmith - Both of these issues have been resolved with new features. Setting the organize_cloud property to true and setting cut_angle to 6.28318530718 (2 × π) should make one point cloud equal one rotation and organize the cloud. In theory, this should make point index 0 always be the first point in each rotation. Let me know if this works for you!