ros-drivers/velodyne #230: No interpolation of azimuth in non-VLP16 unpack() function

shawnhanna, 2019-03-29:
It looks like there is azimuth velocity interpolation for points in the rawdata::unpack_vlp16() function, but not in rawdata::unpack(). Is there a reason to not have this functionality in both functions?

JWhitleyAStuff, 2019-04-02:
@shawnhanna - I am not really sure. I did not originally develop the driver and don't have this level of knowledge of the sensors. Let's see if @jack-oquin can help here.

n-patiphon, 2019-07-18:
@shawnhanna I'm not 100% sure but I think one of the reasons it is not implemented yet is because it would be too complicated. Since each model has a different firing timing, which is necessary for the azimuth interpolation, you would need to change this timing according to the sensor model. Still, I think it's important to accurately convert data packets to point cloud. I found that unpacking data properly can significantly improve mapping results (my use case). 

I forked this repo a while ago (last year) because I needed to custom some parts of it.  I ended up making something like unpack_vlp16, unpack_vlp32c, unpack_hdl32e, etc. I personally think this is not an elegant way to do it, so I didn't make a PR. If the community is interested and is willing to help with my messy source codes, I will work on it when I have time.

kaarta-SHanna, 2019-08-05:
@n-patiphon Could I see the code you wrote that does this? I can collaborate with you to clean it up and return it to the community when we're finished.

JWhitleyWork, 2026-11-16:
[written by AI]

Audit update (2026-11-16): Static review of ROS1 `master` and the current ROS2 `ros2` branch confirms that `unpack()` dispatches VLP-16 to a timing-interpolated path, while the generic path uses each block's rotation. I did not find a merged change or documented rationale resolving the question. Could you specify the non-VLP-16 model and the accuracy/use case expected? No runtime sensor test was performed.

Source: [audited branch code/documentation](https://github.com/ros-drivers/velodyne/blob/29abd0e1361cb7f5eda451d2b51c35eeca45e0d5/velodyne_pointcloud/src/lib/rawdata.cc#L330-L385).