ros-drivers/velodyne #232: [ERROR] [1554173288.904455221]: DriverNodelet::devicePoll - Failed to poll device.

2amour, 2019-04-02:
Hi, I use the velodyne-16 on ubuntu16.04,  but I got the error:[ERROR] [1554173288.904455221]: DriverNodelet::devicePoll - Failed to poll device.
What's the problem?

JWhitleyAStuff, 2019-04-02:
@2amour There are several possible causes. Could you please answer the following (I'll be adding an Issue Template here in a moment):

Are you able to ping the device?
What is the device IP?
Are you building the driver from source or using the version from the build farm?
If using the version from the build farm, which version (`sudo dpkg -s ros-$ROS_VERSION-velodyne`)?

2amour, 2019-04-03:
Hi @JWhitleyAStuff! Thank you for your reply! 
First, my lidar's ip is 192.168.1.201 and my computer's ip is 192.168.1.200. I can successfully ping my lidar.
Second, I build the driver from source.
Actually, if I run the roslaunch velodyne_points VLP16.launch commond  three times, it worked! And I don't konw why.

2amour, 2019-04-03:
this is the output after I run the roslaunch commond:
a409@a409:~$ roslaunch velodyne_pointcloud VLP16_points.launch 
... logging to /home/a409/.ros/log/98a12966-55af-11e9-a56d-4c028912c74a/roslaunch-a409-6094.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://a409:35315/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /velodyne_nodelet_manager_cloud/calibration: /home/a409/catkin...
 * /velodyne_nodelet_manager_cloud/max_range: 130.0
 * /velodyne_nodelet_manager_cloud/min_range: 0.4
 * /velodyne_nodelet_manager_driver/cut_angle: -0.01
 * /velodyne_nodelet_manager_driver/device_ip: 
 * /velodyne_nodelet_manager_driver/frame_id: velodyne
 * /velodyne_nodelet_manager_driver/model: VLP16
 * /velodyne_nodelet_manager_driver/pcap: 
 * /velodyne_nodelet_manager_driver/port: 2368
 * /velodyne_nodelet_manager_driver/read_fast: False
 * /velodyne_nodelet_manager_driver/read_once: False
 * /velodyne_nodelet_manager_driver/repeat_delay: 0.0
 * /velodyne_nodelet_manager_driver/rpm: 600.0
 * /velodyne_nodelet_manager_laserscan/resolution: 0.007
 * /velodyne_nodelet_manager_laserscan/ring: -1

NODES
  /
    velodyne_nodelet_manager (nodelet/nodelet)
    velodyne_nodelet_manager_cloud (nodelet/nodelet)
    velodyne_nodelet_manager_driver (nodelet/nodelet)
    velodyne_nodelet_manager_laserscan (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [6104]
ROS_MASTER_URI=http://a409:11311

setting /run_id to 98a12966-55af-11e9-a56d-4c028912c74a
process[rosout-1]: started with pid [6117]
started core service [/rosout]
process[velodyne_nodelet_manager-2]: started with pid [6130]
process[velodyne_nodelet_manager_driver-3]: started with pid [6135]
process[velodyne_nodelet_manager_cloud-4]: started with pid [6136]
process[velodyne_nodelet_manager_laserscan-5]: started with pid [6141]
[ERROR] [1554254831.996871676]: DriverNodelet::devicePoll - Failed to poll device.
[ERROR] [1554254832.999183912]: DriverNodelet::devicePoll - Failed to poll device.
[ERROR] [1554254832.999183912]: DriverNodelet::devicePoll - Failed to poll device.
[ERROR] [1554254832.999183912]: DriverNodelet::devicePoll - Failed to poll device.
........

jeff-o, 2019-04-18:
I am seeing the same error as 2amour, using a VLP16 Puck running the latest Velodyne firmware, and built from source.

If I use the version from the build farm (1.5.2-0xenial-20190320-193343-0800) installed via apt-get, then it starts without the poll error but fails to publish any data.

2amour, 2019-04-25:
I have solved the problem several days ago! I  set  the value:static const int POLL_TIMEOUT = -1, and it worked!

JWhitleyAStuff, 2019-04-25:
@jeff-o and @2amour - I'm looking into this now. The error that you're seeing is indicative of getting an incomplete scan or a timeout. Can both of you run the same launch file with `--screen` at the end of your launch command and provide me with the output? Thanks for your help.

2amour, 2019-04-25:
Hi, @JWhitleyAStuff I have changed the const int POLL_TIMEOUT = -1, and it worked!
I add the --screen and the output is:
roslaunch velodyne_pointcloud VLP16_points.launch --screen 
... logging to /home/a409/.ros/log/a21acdd0-6761-11e9-8cb1-4c028912c74a/roslaunch-a409-2221.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://a409:34409/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /velodyne_nodelet_manager_cloud/calibration: /home/a409/catkin...
 * /velodyne_nodelet_manager_cloud/max_range: 130.0
 * /velodyne_nodelet_manager_cloud/min_range: 0.4
 * /velodyne_nodelet_manager_driver/cut_angle: -0.01
 * /velodyne_nodelet_manager_driver/device_ip: 
 * /velodyne_nodelet_manager_driver/frame_id: velodyne
 * /velodyne_nodelet_manager_driver/model: VLP16
 * /velodyne_nodelet_manager_driver/pcap: 
 * /velodyne_nodelet_manager_driver/port: 2368
 * /velodyne_nodelet_manager_driver/read_fast: False
 * /velodyne_nodelet_manager_driver/read_once: False
 * /velodyne_nodelet_manager_driver/repeat_delay: 0.0
 * /velodyne_nodelet_manager_driver/rpm: 600.0
 * /velodyne_nodelet_manager_laserscan/resolution: 0.007
 * /velodyne_nodelet_manager_laserscan/ring: -1

NODES
  /
    velodyne_nodelet_manager (nodelet/nodelet)
    velodyne_nodelet_manager_cloud (nodelet/nodelet)
    velodyne_nodelet_manager_driver (nodelet/nodelet)
    velodyne_nodelet_manager_laserscan (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [2232]
ROS_MASTER_URI=http://a409:11311

setting /run_id to a21acdd0-6761-11e9-8cb1-4c028912c74a
process[rosout-1]: started with pid [2245]
started core service [/rosout]
process[velodyne_nodelet_manager-2]: started with pid [2248]
process[velodyne_nodelet_manager_driver-3]: started with pid [2249]
process[velodyne_nodelet_manager_cloud-4]: started with pid [2255]
process[velodyne_nodelet_manager_laserscan-5]: started with pid [2265]
[ INFO] [1556200467.781980474]: Loading nodelet /velodyne_nodelet_manager_driver of type velodyne_driver/DriverNodelet to manager velodyne_nodelet_manager with the following remappings:
[ INFO] [1556200467.796659930]: Loading nodelet /velodyne_nodelet_manager_cloud of type velodyne_pointcloud/CloudNodelet to manager velodyne_nodelet_manager with the following remappings:
[ INFO] [1556200467.800257123]: waitForService: Service [/velodyne_nodelet_manager/load_nodelet] has not been advertised, waiting...
[ INFO] [1556200467.812256263]: waitForService: Service [/velodyne_nodelet_manager/load_nodelet] has not been advertised, waiting...
[ INFO] [1556200467.814775637]: Loading nodelet /velodyne_nodelet_manager_laserscan of type velodyne_laserscan/LaserScanNodelet to manager velodyne_nodelet_manager with the following remappings:
[ INFO] [1556200467.829261957]: waitForService: Service [/velodyne_nodelet_manager/load_nodelet] has not been advertised, waiting...
[ INFO] [1556200467.859710290]: Initializing nodelet with 4 worker threads.
[ INFO] [1556200467.877360011]: waitForService: Service [/velodyne_nodelet_manager/load_nodelet] is now available.
[ INFO] [1556200467.885346586]: waitForService: Service [/velodyne_nodelet_manager/load_nodelet] is now available.
[ INFO] [1556200467.889629362]: waitForService: Service [/velodyne_nodelet_manager/load_nodelet] is now available.
[ INFO] [1556200467.975475350]: Velodyne VLP-16 rotating at 600 RPM
[ INFO] [1556200467.980997041]: publishing 76 packets per scan
[ INFO] [1556200467.998220279]: Cut at specific angle feature deactivated.
[ INFO] [1556200468.029326571]: Reconfigure Request
[ INFO] [1556200468.033772016]: expected frequency: 9.921 (Hz)
[ INFO] [1556200468.043346544]: Opening UDP socket: port 2368
[ INFO] [1556200468.187698150]: correction angles: /home/a409/catkin_ws/src/velodyne/velodyne_pointcloud/params/VLP16db.yaml
[ INFO] [1556200468.201877071]: laser_ring[ 0] =  0, angle = -0.261799
[ INFO] [1556200468.201966433]: laser_ring[ 2] =  1, angle = -0.226893
[ INFO] [1556200468.202021116]: laser_ring[ 4] =  2, angle = -0.191986
[ INFO] [1556200468.202073891]: laser_ring[ 6] =  3, angle = -0.157080
[ INFO] [1556200468.202140526]: laser_ring[ 8] =  4, angle = -0.122173
[ INFO] [1556200468.202193661]: laser_ring[10] =  5, angle = -0.087266
[ INFO] [1556200468.202245872]: laser_ring[12] =  6, angle = -0.052360
[ INFO] [1556200468.202298575]: laser_ring[14] =  7, angle = -0.017453
[ INFO] [1556200468.202586293]: laser_ring[ 1] =  8, angle = +0.017453
[ INFO] [1556200468.202640352]: laser_ring[ 3] =  9, angle = +0.052360
[ INFO] [1556200468.202700819]: laser_ring[ 5] = 10, angle = +0.087266
[ INFO] [1556200468.202753258]: laser_ring[ 7] = 11, angle = +0.122173
[ INFO] [1556200468.202803777]: laser_ring[ 9] = 12, angle = +0.157080
[ INFO] [1556200468.202855580]: laser_ring[11] = 13, angle = +0.191986
[ INFO] [1556200468.202914079]: laser_ring[13] = 14, angle = +0.226893
[ INFO] [1556200468.202964646]: laser_ring[15] = 15, angle = +0.261799
[ INFO] [1556200468.205955887]: Number of lasers: 16.
[ INFO] [1556200468.261400821]: Reconfigure Request

JWhitleyAStuff, 2019-04-25:
@2amour - Changing this value to -1 is not really an option. This essentially means that any given poll of the file descriptor (device) in Linux will wait for an infinite amount of time to read a packet. In theory, this should be OK if the Velodyne continues to send data. However, if no data are sent (for example, the device is turned off), the driver will hang, waiting to receive data.

The current POLL_TIMEOUT value is set for 1 second, which should be plenty of time for the driver to receive a packet so I'm not sure what's happening. Can you change the POLL_TIMEOUT value back to 1000 and run the launch file with the `--screen` again?

doylejg, 2019-05-09:
I have been running into a very similar situation, running with `--screen` options with the following output:
```
started roslaunch server http://lidar-nuc:41607/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.3
 * /velodyne_nodelet_manager_cloud/calibration: /home/lidar-user/...
 * /velodyne_nodelet_manager_cloud/max_range: 130.0
 * /velodyne_nodelet_manager_cloud/min_range: 0.4
 * /velodyne_nodelet_manager_driver/cut_angle: -0.01
 * /velodyne_nodelet_manager_driver/device_ip: 
 * /velodyne_nodelet_manager_driver/frame_id: velodyne
 * /velodyne_nodelet_manager_driver/gps_time: False
 * /velodyne_nodelet_manager_driver/model: VLP16
 * /velodyne_nodelet_manager_driver/pcap: 
 * /velodyne_nodelet_manager_driver/port: 2368
 * /velodyne_nodelet_manager_driver/read_fast: False
 * /velodyne_nodelet_manager_driver/read_once: False
 * /velodyne_nodelet_manager_driver/repeat_delay: 0.0
 * /velodyne_nodelet_manager_driver/rpm: 600.0
 * /velodyne_nodelet_manager_laserscan/resolution: 0.007
 * /velodyne_nodelet_manager_laserscan/ring: -1

NODES
  /
    velodyne_nodelet_manager (nodelet/nodelet)
    velodyne_nodelet_manager_cloud (nodelet/nodelet)
    velodyne_nodelet_manager_driver (nodelet/nodelet)
    velodyne_nodelet_manager_laserscan (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

process[velodyne_nodelet_manager-1]: started with pid [3798]
process[velodyne_nodelet_manager_driver-2]: started with pid [3799]
process[velodyne_nodelet_manager_cloud-3]: started with pid [3800]
process[velodyne_nodelet_manager_laserscan-4]: started with pid [3801]
[ INFO] [1557441704.693276756]: Loading nodelet /velodyne_nodelet_manager_cloud of type velodyne_pointcloud/CloudNodelet to manager velodyne_nodelet_manager with the following remappings:
[ INFO] [1557441704.694749876]: waitForService: Service [/velodyne_nodelet_manager/load_nodelet] has not been advertised, waiting...
[ INFO] [1557441704.702657328]: Loading nodelet /velodyne_nodelet_manager_driver of type velodyne_driver/DriverNodelet to manager velodyne_nodelet_manager with the following remappings:
[ INFO] [1557441704.704226835]: Loading nodelet /velodyne_nodelet_manager_laserscan of type velodyne_laserscan/LaserScanNodelet to manager velodyne_nodelet_manager with the following remappings:
[ INFO] [1557441704.705113254]: waitForService: Service [/velodyne_nodelet_manager/load_nodelet] has not been advertised, waiting...
[ INFO] [1557441704.705919494]: Initializing nodelet with 4 worker threads.
[ INFO] [1557441704.715473004]: waitForService: Service [/velodyne_nodelet_manager/load_nodelet] is now available.
[ INFO] [1557441704.718645817]: correction angles: /home/lidar-user/catkin_ws/src/velodyne/velodyne_pointcloud/params/VLP16db.yaml
[ INFO] [1557441704.720943088]: laser_ring[ 0] =  0, angle = -0.261799
[ INFO] [1557441704.720961648]: laser_ring[ 2] =  1, angle = -0.226893
[ INFO] [1557441704.720972149]: laser_ring[ 4] =  2, angle = -0.191986
[ INFO] [1557441704.720986484]: laser_ring[ 6] =  3, angle = -0.157080
[ INFO] [1557441704.720998261]: laser_ring[ 8] =  4, angle = -0.122173
[ INFO] [1557441704.721010526]: laser_ring[10] =  5, angle = -0.087266
[ INFO] [1557441704.721022769]: laser_ring[12] =  6, angle = -0.052360
[ INFO] [1557441704.721038150]: laser_ring[14] =  7, angle = -0.017453
[ INFO] [1557441704.721060970]: laser_ring[ 1] =  8, angle = +0.017453
[ INFO] [1557441704.721076442]: laser_ring[ 3] =  9, angle = +0.052360
[ INFO] [1557441704.721090987]: laser_ring[ 5] = 10, angle = +0.087266
[ INFO] [1557441704.721105441]: laser_ring[ 7] = 11, angle = +0.122173
[ INFO] [1557441704.721119149]: laser_ring[ 9] = 12, angle = +0.157080
[ INFO] [1557441704.721133812]: laser_ring[11] = 13, angle = +0.191986
[ INFO] [1557441704.721145891]: laser_ring[13] = 14, angle = +0.226893
[ INFO] [1557441704.721157448]: laser_ring[15] = 15, angle = +0.261799
[ INFO] [1557441704.721278698]: Number of lasers: 16.
[ INFO] [1557441704.725829080]: waitForService: Service [/velodyne_nodelet_manager/load_nodelet] is now available.
[ INFO] [1557441704.726562048]: Reconfigure Request
[ INFO] [1557441704.733695562]: Velodyne VLP-16 rotating at 600 RPM
[ INFO] [1557441704.733922823]: publishing 76 packets per scan
[ INFO] [1557441704.734662775]: Cut at specific angle feature deactivated.
[ INFO] [1557441704.736757632]: Reconfigure Request
[ INFO] [1557441704.737066919]: expected frequency: 9.921 (Hz)
[ INFO] [1557441704.737933786]: Opening UDP socket: port 2368
[ WARN] [1557441705.739138096]: Velodyne poll() timeout
[ERROR] [1557441705.739336453]: DriverNodelet::devicePoll - Failed to poll device.
[ WARN] [1557441706.740661158]: Velodyne poll() timeout
[ERROR] [1557441706.740778421]: DriverNodelet::devicePoll - Failed to poll device.
```

Hershey988, 2019-05-15:
Do you have any solutions for this yet?

JWhitleyAStuff, 2019-05-15:
@2amour / @doylejg / @Hershey988 - Can you please clone from source and see if you still have this issue? We have added quite a bit to the driver since 1.5.2 (January release) and the current `master` needs a good amount of testing anyway.

jeff-o, 2019-05-15:
I just tried again today with a brand-new Velodyne direct from the manufacturer. I get the same pattern as before. I am running 16.04 Kinetic:

When I install from debs, then the driver launches without errors but fails to publish any data to /velodyne_points.

When I run from source, the driver launches with the "Failed to poll device" error.

jeff-o, 2019-05-15:
administrator@CPR-J100-0309:/etc/ros/kinetic/ros.d$ roslaunch VLP16_points.launch --screen
... logging to /home/administrator/.ros/log/88e03496-772d-11e9-95a2-c400ad2240b7/roslaunch-CPR-J100-0309-16668.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://CPR-J100-0309:44062/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /velodyne_nodelet_manager_cloud/calibration: /home/administrat...
 * /velodyne_nodelet_manager_cloud/max_range: 130.0
 * /velodyne_nodelet_manager_cloud/min_range: 0.4
 * /velodyne_nodelet_manager_driver/cut_angle: -0.01
 * /velodyne_nodelet_manager_driver/device_ip: 192.168.131.20
 * /velodyne_nodelet_manager_driver/frame_id: velodyne
 * /velodyne_nodelet_manager_driver/gps_time: False
 * /velodyne_nodelet_manager_driver/model: VLP16
 * /velodyne_nodelet_manager_driver/pcap: 
 * /velodyne_nodelet_manager_driver/port: 2368
 * /velodyne_nodelet_manager_driver/read_fast: False
 * /velodyne_nodelet_manager_driver/read_once: False
 * /velodyne_nodelet_manager_driver/repeat_delay: 0.0
 * /velodyne_nodelet_manager_driver/rpm: 600.0
 * /velodyne_nodelet_manager_laserscan/resolution: 0.007
 * /velodyne_nodelet_manager_laserscan/ring: -1

NODES
  /
    velodyne_nodelet_manager (nodelet/nodelet)
    velodyne_nodelet_manager_cloud (nodelet/nodelet)
    velodyne_nodelet_manager_driver (nodelet/nodelet)
    velodyne_nodelet_manager_laserscan (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [16687]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 88e03496-772d-11e9-95a2-c400ad2240b7
process[rosout-1]: started with pid [16710]
started core service [/rosout]
process[velodyne_nodelet_manager-2]: started with pid [16734]
process[velodyne_nodelet_manager_driver-3]: started with pid [16744]
process[velodyne_nodelet_manager_cloud-4]: started with pid [16747]
process[velodyne_nodelet_manager_laserscan-5]: started with pid [16748]
[ INFO] [1557937307.265156151]: Loading nodelet /velodyne_nodelet_manager_driver of type velodyne_driver/DriverNodelet to manager velodyne_nodelet_manager with the following remappings:
[ INFO] [1557937307.271137528]: Loading nodelet /velodyne_nodelet_manager_cloud of type velodyne_pointcloud/CloudNodelet to manager velodyne_nodelet_manager with the following remappings:
[ INFO] [1557937307.274935376]: waitForService: Service [/velodyne_nodelet_manager/load_nodelet] has not been advertised, waiting...
[ INFO] [1557937307.277391465]: Loading nodelet /velodyne_nodelet_manager_laserscan of type velodyne_laserscan/LaserScanNodelet to manager velodyne_nodelet_manager with the following remappings:
[ INFO] [1557937307.278792593]: waitForService: Service [/velodyne_nodelet_manager/load_nodelet] has not been advertised, waiting...
[ INFO] [1557937307.279696875]: Initializing nodelet with 4 worker threads.
[ INFO] [1557937307.297104204]: waitForService: Service [/velodyne_nodelet_manager/load_nodelet] is now available.
[ INFO] [1557937307.301019815]: waitForService: Service [/velodyne_nodelet_manager/load_nodelet] is now available.
[ INFO] [1557937307.334617671]: Velodyne VLP-16 rotating at 600 RPM
[ INFO] [1557937307.336036588]: publishing 76 packets per scan
[ INFO] [1557937307.341473005]: Cut at specific angle feature deactivated.
[ INFO] [1557937307.352272078]: Reconfigure Request
[ INFO] [1557937307.353829256]: expected frequency: 9.921 (Hz)
[ INFO] [1557937307.359897580]: Only accepting packets from IP address: 192.168.131.20
[ INFO] [1557937307.359978913]: Opening UDP socket: port 2368
[ INFO] [1557937307.372317456]: correction angles: /home/administrator/catkin_ws/src/velodyne/velodyne_pointcloud/params/VLP16db.yaml
[ INFO] [1557937307.376357769]: laser_ring[ 0] =  0, angle = -0.261799
[ INFO] [1557937307.376387339]: laser_ring[ 2] =  1, angle = -0.226893
[ INFO] [1557937307.376412011]: laser_ring[ 4] =  2, angle = -0.191986
[ INFO] [1557937307.376430934]: laser_ring[ 6] =  3, angle = -0.157080
[ INFO] [1557937307.376454859]: laser_ring[ 8] =  4, angle = -0.122173
[ INFO] [1557937307.376478063]: laser_ring[10] =  5, angle = -0.087266
[ INFO] [1557937307.376500923]: laser_ring[12] =  6, angle = -0.052360
[ INFO] [1557937307.376522336]: laser_ring[14] =  7, angle = -0.017453
[ INFO] [1557937307.376544730]: laser_ring[ 1] =  8, angle = +0.017453
[ INFO] [1557937307.376565194]: laser_ring[ 3] =  9, angle = +0.052360
[ INFO] [1557937307.376586376]: laser_ring[ 5] = 10, angle = +0.087266
[ INFO] [1557937307.376606438]: laser_ring[ 7] = 11, angle = +0.122173
[ INFO] [1557937307.376627226]: laser_ring[ 9] = 12, angle = +0.157080
[ INFO] [1557937307.376646750]: laser_ring[11] = 13, angle = +0.191986
[ INFO] [1557937307.376669317]: laser_ring[13] = 14, angle = +0.226893
[ INFO] [1557937307.376688807]: laser_ring[15] = 15, angle = +0.261799
[ INFO] [1557937307.376905424]: Number of lasers: 16.
[ INFO] [1557937307.396890552]: Reconfigure Request
[ WARN] [1557937308.362975279]: Velodyne poll() timeout
[ERROR] [1557937308.363089149]: DriverNodelet::devicePoll - Failed to poll device.
[ WARN] [1557937309.364288978]: Velodyne poll() timeout
[ERROR] [1557937309.364583402]: DriverNodelet::devicePoll - Failed to poll device.
[ WARN] [1557937310.365920538]: Velodyne poll() timeout
[ERROR] [1557937310.366061011]: DriverNodelet::devicePoll - Failed to poll device.

JWhitleyAStuff, 2019-05-15:
@jeff-o I'm guessing the last log is running from source?

JWhitleyAStuff, 2019-05-15:
Also, it looks like this is strictly happening on VLP-16s. Has anyone here tested with anything other than a VLP-16? We have not been seeing this issue in our office with VLP-32s.

jeff-o, 2019-05-15:
@JWhitleyAStuff yes, that's running from source. Here it is, running from debs. Notice it still complains about a Velodyne poll() timeout:

administrator@CPR-J100-0309:/etc/ros/kinetic/ros.d$ roslaunch VLP16_points.launch --screen
... logging to /home/administrator/.ros/log/c27517e2-7734-11e9-95a2-c400ad2240b7/roslaunch-CPR-J100-0309-18327.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://CPR-J100-0309:34027/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /velodyne_nodelet_manager_cloud/calibration: /opt/ros/kinetic/...
 * /velodyne_nodelet_manager_cloud/max_range: 130.0
 * /velodyne_nodelet_manager_cloud/min_range: 0.4
 * /velodyne_nodelet_manager_driver/cut_angle: -0.01
 * /velodyne_nodelet_manager_driver/device_ip: 192.168.131.20
 * /velodyne_nodelet_manager_driver/frame_id: velodyne
 * /velodyne_nodelet_manager_driver/model: VLP16
 * /velodyne_nodelet_manager_driver/pcap: 
 * /velodyne_nodelet_manager_driver/port: 2368
 * /velodyne_nodelet_manager_driver/read_fast: False
 * /velodyne_nodelet_manager_driver/read_once: False
 * /velodyne_nodelet_manager_driver/repeat_delay: 0.0
 * /velodyne_nodelet_manager_driver/rpm: 600.0
 * /velodyne_nodelet_manager_laserscan/resolution: 0.007
 * /velodyne_nodelet_manager_laserscan/ring: -1

NODES
  /
    velodyne_nodelet_manager (nodelet/nodelet)
    velodyne_nodelet_manager_cloud (nodelet/nodelet)
    velodyne_nodelet_manager_driver (nodelet/nodelet)
    velodyne_nodelet_manager_laserscan (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [18346]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to c27517e2-7734-11e9-95a2-c400ad2240b7
process[rosout-1]: started with pid [18369]
started core service [/rosout]
process[velodyne_nodelet_manager-2]: started with pid [18394]
process[velodyne_nodelet_manager_driver-3]: started with pid [18402]
process[velodyne_nodelet_manager_cloud-4]: started with pid [18405]
process[velodyne_nodelet_manager_laserscan-5]: started with pid [18407]
[ INFO] [1557940410.333637981]: Loading nodelet /velodyne_nodelet_manager_driver of type velodyne_driver/DriverNodelet to manager velodyne_nodelet_manager with the following remappings:
[ INFO] [1557940410.340600687]: Initializing nodelet with 4 worker threads.
[ INFO] [1557940410.341213493]: Loading nodelet /velodyne_nodelet_manager_cloud of type velodyne_pointcloud/CloudNodelet to manager velodyne_nodelet_manager with the following remappings:
[ INFO] [1557940410.342439515]: waitForService: Service [/velodyne_nodelet_manager/load_nodelet] has not been advertised, waiting...
[ INFO] [1557940410.345879344]: Loading nodelet /velodyne_nodelet_manager_laserscan of type velodyne_laserscan/LaserScanNodelet to manager velodyne_nodelet_manager with the following remappings:
[ INFO] [1557940410.356900779]: correction angles: /opt/ros/kinetic/share/velodyne_pointcloud/params/VLP16db.yaml
[ INFO] [1557940410.362814836]: laser_ring[ 0] =  0, angle = -0.261799
[ INFO] [1557940410.362885419]: laser_ring[ 2] =  1, angle = -0.226893
[ INFO] [1557940410.362947433]: laser_ring[ 4] =  2, angle = -0.191986
[ INFO] [1557940410.362995504]: laser_ring[ 6] =  3, angle = -0.157080
[ INFO] [1557940410.363059447]: laser_ring[ 8] =  4, angle = -0.122173
[ INFO] [1557940410.363120873]: laser_ring[10] =  5, angle = -0.087266
[ INFO] [1557940410.363182996]: laser_ring[12] =  6, angle = -0.052360
[ INFO] [1557940410.363233251]: laser_ring[14] =  7, angle = -0.017453
[ INFO] [1557940410.363286287]: laser_ring[ 1] =  8, angle = +0.017453
[ INFO] [1557940410.363344757]: laser_ring[ 3] =  9, angle = +0.052360
[ INFO] [1557940410.363406921]: laser_ring[ 5] = 10, angle = +0.087266
[ INFO] [1557940410.363465334]: laser_ring[ 7] = 11, angle = +0.122173
[ INFO] [1557940410.363519757]: laser_ring[ 9] = 12, angle = +0.157080
[ INFO] [1557940410.363583102]: laser_ring[11] = 13, angle = +0.191986
[ INFO] [1557940410.363644624]: laser_ring[13] = 14, angle = +0.226893
[ INFO] [1557940410.363697261]: laser_ring[15] = 15, angle = +0.261799
[ INFO] [1557940410.363942448]: Number of lasers: 16.
[ INFO] [1557940410.364723826]: waitForService: Service [/velodyne_nodelet_manager/load_nodelet] is now available.
[ INFO] [1557940410.382772708]: Reconfigure Request
[ INFO] [1557940410.452500417]: Velodyne VLP-16 rotating at 600 RPM
[ INFO] [1557940410.453904040]: publishing 76 packets per scan
[ INFO] [1557940410.459307053]: Cut at specific angle feature deactivated.
[ INFO] [1557940410.469883849]: Reconfigure Request
[ INFO] [1557940410.471506447]: expected frequency: 9.921 (Hz)
[ INFO] [1557940410.474362572]: Only accepting packets from IP address: 192.168.131.20
[ INFO] [1557940410.474404320]: Opening UDP socket: port 2368
[ WARN] [1557940411.477214123]: Velodyne poll() timeout

jeff-o, 2019-05-15:
The above tests were done using the VLP16 3.0.40 firmware. I pushed it back to 3.0.38 and tried again, but results were the same.

jeff-o, 2019-05-15:
More results, and apologies for the multiple messages. 

When the firmware change made no difference, I returned everything to defaults (no host destination, factory-default IP). I was able to view the data in VeloView like that (aside: why doesn't VeloView allow you to specify the device IP??) After confirming that data WAS being produced, I set up the robot to access the VLP16 at the default 192.168.1.201. No luck, I still got the timeouts. So, I switched everything back to how I had it at the beginning, with the VLP16 set for a specific Host IP and a different IP address (192.168.131.20). Connected everything up again, and lo and behold - the darned thing starts publishing data in ROS! It works whether I use debs or built from source.

All I can surmise is that something on the VLP16 isn't being set properly with regards to the communications settings, and that it may take multiple attempts to make the settings "stick." Can anyone else try changing their VLP16's IP address, perhaps multiple times, to see if the problem goes away?

JWhitleyAStuff, 2019-05-15:
@jeff-o I'm glad to hear it's working! I wonder if it is related to the Host IP in the Velodyne. Was it set to 255.255.255.255 before?

jeff-o, 2019-05-15:
When we configure it for our robots, we set the host IP to that of the robot, to avoid spamming the entire robot network with velodyne data. But yes, from the factory the host IP was 255.255.255.255.

Hershey988, 2019-05-15:
> More results, and apologies for the multiple messages.
>
> When the firmware change made no difference, I returned everything to defaults (no host destination, factory-default IP). I was able to view [...]

What is your Host IP set as?
What is your Network(sensor) IP set as?

jeff-o, 2019-05-15:
Currently:

Host (Destination) IP Address: 192.168.131.1
Network (Sensor) IP Address: 192.168.131.20
Network (Sensor) Gateway: 192.168.131.1

The IP address of the robot is, of course, 192.168.131.1.

Hershey988, 2019-05-15:
@jeff-o I have used your configuration and I am having no luck. Here are the IPs I am using.

Host (Destination) IP Address: 192.168.1.1
Network (Sensor) IP Address: 192.168.1.201
Network (Sensor) Gateway: 192.168.1.1

I have also tried other combinations, but I am having no luck.

JWhitleyAStuff, 2019-05-15:
@jeff-o and @Hershey988 - I do not recommend setting a gateway on either your host device's interface or the Velodyne. These devices do not need access to any other networks (what a gateway does) so set them both to have a gateway of 0.0.0.0. Also, if you set the "Host IP Address" to 192.168.1.1, make sure your PC actually has that IP address. In addition, I would recommend the subnet of 255.255.255.0.

@Hershey988 - Would you mind posting a screenshot of your Velodyne web interface and your IP configuration from Ubuntu for the Ethernet port that connects to the Velodyne?

Hershey988, 2019-05-15:
[image]
[image]

Kennyal3, 2019-05-16:
I'm having this issue as well with the VLP-32C. I am able to connect to the web interface just fine as well as successfully ping the device. Using ROS Kinetic on Ubuntu16.04.

JWhitleyAStuff, 2019-05-17:
@Kennyal3 and others still having trouble - If you're an AS customer, please send your request to support@autonomoustuff.com. If not, please create a new issue. This appears to have just been an IP configuration problem.

closed 2019-05-17 by JWhitleyAStuff

Kennyal3, 2019-05-17:
Hi Josh,
This is pretty embarrassing but the problem was me, the user. Not taking
advantage of roslaunch input args and the device parameters were incorrect.
It's working great now.

teomandi, 2019-09-18:
[Solved] Hi all.
I had the same problem for 2 days. I coundn't get the velodyne's data. The problem was that the UDP communication is very insecure, the firewall does not accept the came data. So,
I disabled the firewall using `sudo ufw disable` and it worked.

tangshp, 2019-10-28:
I find that the data port(2356) in launch is different to the device data port(3201).

DanielArnett, 2019-10-28:
My solution is to either kill and restart the velodyne node (works with 1.5.2 and master), or send the bash command 
```
curl --data "rpm=600" http://10.1.12.201/cgi/setting; sleep 1; curl --data "laser=on" http://10.1.12.201/cgi/setting
```
before launching (working with master, not 1.5.2).

chinhyuntai, 2019-10-31:
I also had same problem on melodic with VLP16 in master. Port and IP was set correctly, but it was showing the error message like 
[ WARN] [1557937310.365920538]: Velodyne poll() timeout
[ERROR] [1557937310.366061011]: DriverNodelet::devicePoll - Failed to poll device.

My solution was using 1.5.2 instead of master, and it doesn't show the message now.

waalbukhanajer, 2022-05-05:
In my case, I solved the issue by making the destination IP address similar to what in the Velodyne Ultra Puck VPL-32C setup.
e.g., The Destination IP: 192.168.1.10
The laptop eth0 IP was set to: 192.168.1.10 also (instead to anything else 192.168.1.xx)

I am using
ROS Noetic, 
Linux Ubuntu 20.04 Focal
Host (Destination) IP Address: 192.168.1.10
Network (Sensor) IP Address: 192.168.1.201
Network (Sensor) Gateway: 192.168.1.1
Host (Laptop-Host) IP Address: 192.168.1.10
[image]