ros-drivers/velodyne #233: C++ 11 or greater

kmhallen, 2019-04-04:
It looks like the C++ standard was set to C++11 recently. Indigo/Trusty and Kinetic/Xenial default to C++98, so this is necessary. However, Melodic/Bionic defaults to the newer C++14, and reducing the standard to C++11 may be undesired.

The following cmake lines set the C++ standard to C++11 only when necessary. This has been tested in several Dataspeed packages for ROS Indigo/Kinetic/Melodic for Ubuntu.

```cmake
if (NOT "${CMAKE_CXX_STANDARD_COMPUTED_DEFAULT}")
  message(STATUS "Changing CXX_STANDARD from C++98 to C++11")
  set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")
elseif ("${CMAKE_CXX_STANDARD_COMPUTED_DEFAULT}" STREQUAL "98")
  message(STATUS "Changing CXX_STANDARD from C++98 to C++11")
  set(CMAKE_CXX_STANDARD 11)
endif()
```

JWhitleyAStuff, 2019-04-08:
@kmhallen - Thanks for the info. Would you mind making a PR (I've just been given access to maintain `image_pipeline` and it's taking up a lot of my time at the moment)? Thanks!

kmhallen, 2019-04-08:
PR #234

JWhitleyAStuff, 2019-04-11:
Closed by #234

closed 2019-04-11 by JWhitleyAStuff