ros-drivers/velodyne #240: why does my VLP-16 have 20° vertical field of view?

aybakana, 2019-05-06:
vert_correction values of the calibration file(VLP-16.yaml) are:
--
-0.17453292519943295
0.011641346110802179
-0.15126768627034853
0.03490658503988659
-0.1279849940487442
0.058171823968971005
-0.10471975511965978
0.08145451619057535
-0.08145451619057535
0.10471975511965978
-0.058171823968971005
0.1279849940487442
-0.03490658503988659
0.15126768627034853
-0.011641346110802179
0.17453292519943295

After I went through the code inside "rawdata.cc", I figured out that vert_correction value is used as the angle of the laser ray. But when I convert my vert_correction values into degrees, I got the angles of 0.667, 2, 3.333, 4.667, 6, 7.333, 8.667, 10. According to the manual, VLP-16 has to have 30° vertical field of view. Has anyone encountered something similar?

maximus5684, 2019-05-08:
@aybakana Are you positive the sensor you have isn't a Puck Hi-Res? The standard Puck (VLP-16) has a vertical field-of-view of 30 degrees (+15/-15) while the Puck Hi-Res has a 20 degree field-of-view (+10/-10). Maybe Velodyne sent you the wrong product or the wrong configuration file?

JWhitleyAStuff, 2019-05-08:
@aybakana Are you positive the sensor you have isn't a Puck Hi-Res? The standard Puck (VLP-16) has a vertical field-of-view of 30 degrees (+15/-15) while the Puck Hi-Res (VLP-16-HI-RES) has a 20 degree field-of-view (+10/-10). Maybe Velodyne sent you the wrong product or the wrong configuration file?

aybakana, 2019-05-10:
Thanks for your reply. We have 2 VLP-16. I was informed that the one is a standart Puck and the other is Puck Lite but both of their calibration files look like they are Puck Hi-Res. I guess there was been a mistake.

closed 2019-05-10 by aybakana