ros-drivers/velodyne #242: Introduce phase_lock_angle param to control msg header timestamp

mpitropov, 2019-05-14:
**Is your feature request related to a problem? Please describe.**
Within the driver the packet message is published using the timestamp of the last packet. This means that the published pointcloud has the timestamp of the last packet. Also the transformed pountcloud is corrected to the timestamp of the last packet.
https://github.com/ros-drivers/velodyne/blob/7ecdf17173a609234df4e61cdd443748ccf7ab8f/velodyne_driver/src/driver/driver.cc#L243
This creates a problem with time sync where someone has a PPS signal triggering other sensors and also has a phase lock angle for their lidar. The lidar will have a delayed timestamp compared to other sensors.

**Describe the solution you'd like**
I think I'll have to create a param called phase_lock_angle which should be the same angle you have set in your lidar configuration. Adding to the current cut_angle loops seems to be the way to go although the double while loop is looking a bit large. When the packet angle passes the phase_lock_angle I will store the packet timestamp to be used for the published message header.

**Describe alternatives you've considered**
I have the cut angle set to behind the car and the phase lock set to in front of the car so in theory I would be able to just grab the timestamp of the center packet. Other people may have different setups though.

**Additional context**
[image]

JWhitleyAStuff, 2019-05-15:
I agree with the general concept. Would you be willing/able to put in a Pull Request with this implemented?

JWhitleyWork, 2026-11-16:
[written by AI]

Audit update (2026-11-16): I checked ROS 1 `master` and current ROS 2 `ros2`; neither has a configurable phase-lock-angle timestamp. [PR #481](https://github.com/ros-drivers/velodyne/pull/481) proposes that ROS 1 behavior but remains open with review follow-up pending. [PR #285](https://github.com/ros-drivers/velodyne/pull/285) is a separate, limited scan-separation proposal, not the requested phase-lock option. No merged PR currently resolves this request.

Source: [audited branch code/documentation](https://github.com/ros-drivers/velodyne/blob/29abd0e1361cb7f5eda451d2b51c35eeca45e0d5/velodyne_driver/src/driver/driver.cc#L216-L280).