ros-drivers/velodyne #244: Velodyne vlp-16 frequency too low

alec-gurman, 2019-05-23:
Hey guys,

Im running a velodyne VLP16, however i am having issues with it's publish frequency, this lidar is spinning at 600 rpm with a packet rate of 754. The /diagnostics topic for the `velodyne_nodelet_manager` currently show's this:

```
values: 
  - 
    key: "Events in window"
    value: "46"
  - 
    key: "Events since startup"
    value: "2774"
  - 
    key: "Duration of window (s)"
    value: "10.884267"
  - 
    key: "Actual frequency (Hz)"
    value: "4.226284"
  - 
    key: "Target frequency (Hz)"
    value: "9.921053"
  - 
    key: "Minimum acceptable frequency (Hz)"
    value: "8.928947"
  - 
    key: "Maximum acceptable frequency (Hz)"
    value: "10.913158"
  - 
    key: "Earliest timestamp delay:"
    value: "0.000624"
  - 
    key: "Latest timestamp delay:"
    value: "0.000779"
  - 
    key: "Earliest acceptable timestamp delay:"
    value: "-1.000000"
  - 
    key: "Latest acceptable timestamp delay:"
    value: "5.000000"
  - 
    key: "Late diagnostic update count:"
    value: "0"
  - 
    key: "Early diagnostic update count:"
    value: "0"
  - 
    key: "Zero seen diagnostic update count:"
    value: "0"
```

You can see that i should be getting a frequency of around 10hz, however the actual frequency is only around 4.1hz. Does anyone have any idea what could be the cause of this?

Here are some more outputs:

`rostopic bw /velodyne_points`

```
average: 3.77MB/s
    mean: 0.90MB min: 0.89MB max: 0.90MB window: 29
```

`rostopic hz /velodyne_points`

```
average rate: 4.196
    min: 0.216s max: 0.276s std dev: 0.02812s window: 20
```

`rostopic bw /velodyne_packets`

```
average: 434.13KB/s
    mean: 92.29KB min: 92.29KB max: 92.29KB window: 9
```

The velodyne_nodelet_manager seems to be using about 0.6% of memory and 2% of cpu, and the computer i am streaming the lidar too does not show any signs of computational burden

I should also note from a hardware layer perspective the lidar has it's own dedicated ethernet port into the pc, (not running through a swtich) so i would suspect there shouldn't be a bandwidth issue?

JWhitleyAStuff, 2019-05-23:
@alec-gurman can you post a screenshot of the web interface for your VLP-16 as well as a copy of the cfg file and launch file that you're using? Thanks.

alec-gurman, 2019-05-23:
Hey @JWhitleyAStuff, cheers for the reply, see below the requested information, most of the configuration paramters are in the launch file itself, and the calibration paramters have a seperate yaml file which i have also included.

**Launch file**

```
<!-- -*- mode: XML -*- -->
<!-- run velodyne_pointcloud/CloudNodelet in a nodelet manager for a VLP-16 -->

<launch>

  <!-- declare arguments with default values -->
  <arg name="calibration" default="$(find velodyne_pointcloud)/params/VLP16db.yaml"/>
  <arg name="device_ip" default="" />
  <arg name="frame_id" default="velodyne" />
  <arg name="manager" default="$(arg frame_id)_nodelet_manager" />
  <arg name="max_range" default="130.0" />
  <arg name="min_range" default="0.4" />
  <arg name="pcap" default="" />
  <arg name="port" default="2368" />
  <arg name="read_fast" default="false" />
  <arg name="read_once" default="false" />
  <arg name="repeat_delay" default="0.0" />
  <arg name="rpm" default="600.0" />
  <arg name="cut_angle" default="-0.01" />
  <arg name="laserscan_ring" default="-1" />
  <arg name="laserscan_resolution" default="0.007" />

  <!-- start nodelet manager and driver nodelets -->
  <include file="$(find velodyne_driver)/launch/nodelet_manager.launch">
    <arg name="device_ip" value="$(arg device_ip)"/>
    <arg name="frame_id" value="$(arg frame_id)"/>
    <arg name="manager" value="$(arg manager)" />
    <arg name="model" value="VLP16"/>
    <arg name="pcap" value="$(arg pcap)"/>
    <arg name="port" value="$(arg port)"/>
    <arg name="read_fast" value="$(arg read_fast)"/>
    <arg name="read_once" value="$(arg read_once)"/>
    <arg name="repeat_delay" value="$(arg repeat_delay)"/>
    <arg name="rpm" value="$(arg rpm)"/>
    <arg name="cut_angle" value="$(arg cut_angle)"/>
  </include>

  <!-- start cloud nodelet -->
  <include file="$(find velodyne_pointcloud)/launch/cloud_nodelet.launch">
    <arg name="calibration" value="$(arg calibration)"/>
    <arg name="manager" value="$(arg manager)" />
    <arg name="max_range" value="$(arg max_range)"/>
    <arg name="min_range" value="$(arg min_range)"/>
  </include>

  <!-- start laserscan nodelet -->
  <include file="$(find velodyne_pointcloud)/launch/laserscan_nodelet.launch">
    <arg name="manager" value="$(arg manager)" />
    <arg name="ring" value="$(arg laserscan_ring)"/>
    <arg name="resolution" value="$(arg laserscan_resolution)"/>
  </include>

</launch>
```

**Calibration file**

```
lasers:
- {dist_correction: 0.0, dist_correction_x: 0.0, dist_correction_y: 0.0, focal_distance: 0.0,
  focal_slope: 0.0, horiz_offset_correction: 0.0, laser_id: 0, rot_correction: 0.0,
  vert_correction: -0.2617993877991494, vert_offset_correction: 0.0}
- {dist_correction: 0.0, dist_correction_x: 0.0, dist_correction_y: 0.0, focal_distance: 0.0,
  focal_slope: 0.0, horiz_offset_correction: 0.0, laser_id: 1, rot_correction: 0.0,
  vert_correction: 0.017453292519943295, vert_offset_correction: 0.0}
[... 26 lines cut ...]
  vert_correction: -0.08726646259971647, vert_offset_correction: 0.0}
- {dist_correction: 0.0, dist_correction_x: 0.0, dist_correction_y: 0.0, focal_distance: 0.0,
  focal_slope: 0.0, horiz_offset_correction: 0.0, laser_id: 11, rot_correction: 0.0,
  vert_correction: 0.19198621771937624, vert_offset_correction: 0.0}
- {dist_correction: 0.0, dist_correction_x: 0.0, dist_correction_y: 0.0, focal_distance: 0.0,
  focal_slope: 0.0, horiz_offset_correction: 0.0, laser_id: 12, rot_correction: 0.0,
  vert_correction: -0.05235987755982989, vert_offset_correction: 0.0}
- {dist_correction: 0.0, dist_correction_x: 0.0, dist_correction_y: 0.0, focal_distance: 0.0,
  focal_slope: 0.0, horiz_offset_correction: 0.0, laser_id: 13, rot_correction: 0.0,
  vert_correction: 0.22689280275926285, vert_offset_correction: 0.0}
- {dist_correction: 0.0, dist_correction_x: 0.0, dist_correction_y: 0.0, focal_distance: 0.0,
  focal_slope: 0.0, horiz_offset_correction: 0.0, laser_id: 14, rot_correction: 0.0,
  vert_correction: -0.017453292519943295, vert_offset_correction: 0.0}
- {dist_correction: 0.0, dist_correction_x: 0.0, dist_correction_y: 0.0, focal_distance: 0.0,
  focal_slope: 0.0, horiz_offset_correction: 0.0, laser_id: 15, rot_correction: 0.0,
  vert_correction: 0.2617993877991494, vert_offset_correction: 0.0}
num_lasers: 16
distance_resolution: 0.002
```

**Web interface screenshot**

[image]
[image]
[image]

alec-gurman, 2019-05-28:
I managed to "resolve" this issue. In the configuration web interface for the VLP-16 i was originally making use of the FOV Start parameter to crop the data between two angles. Setting this range back to 0 to 359 restored to the 10hz frequency. Is this a possible bug?

JWhitleyAStuff, 2019-05-28:
@alec-gurman - This is what I was going to suggest. I believe this is not a "bug," per-se, just a limitation of the current design. The driver assumes a certain number of packets per revolution for a given model. If that number is not met, it will wait until the data container is full before publishing. If you check out https://github.com/ros-drivers/velodyne/blob/fb53a02d5a011a2858ff192c2dee98200f0147af/velodyne_driver/src/driver/driver.cc#L66, you will see what I'm talking about. While the RPM is configurable (to match that set on the sensor), the number of expected packets per revolution is currently not configurable (the `npackets` value). However, this is a feature that has been requested before. As such, I've created #245 to address the issue. That being said, I don't know if I'll be able to work on it any time soon and any help would be much appreciated.

JWhitleyAStuff, 2019-05-28:
Please also note that the number of packets per revolution is not steady given a slight variance in the revolutions per minute of the sensor. To combat this, you can enable the "cut at specified angle" feature by setting the "cut_angle" in radians. Setting this to, for example, 6.28318530718 (2 * pi) will tell the driver that you always want 1 point cloud = 1 rotation. This feature *may* be able to be used to limit the number of packets when

JWhitleyAStuff, 2019-05-28:
@alec-gurman - I've been digging a little deeper into the "cut_at_specified_angle" feature that was added a few months ago. Try setting the `cut_angle` parameter in your .cfg file to 3.66519 (this is 285 - 75 degrees in radians). I'm not positive that this will work (it may assume that the start has to be 0) but it's worth a try.

closed 2019-06-14 by JWhitleyAStuff