ros-drivers/velodyne #245: Add Configurable Packets/Revolution Value

JWhitleyAStuff, 2019-05-28:
**Is your feature request related to a problem? Please describe.**
When the Velodyne is configured to use a custom start or end FOV, the default packet rate is not acceptable.

**Describe the solution you'd like**
Add one of two features:

1. Number of packets per revolution is configurable by the user.
2. The start/end FOV parameters can be input by the user and the number of packets per revolution is calculated automatically.

**Additional context**
This has been suggested before in #244, #33, and others.

JWhitleyAStuff, 2019-05-28:
I'm not positive, but the `cut_at_specified_angle` feature may fix this. Testing...

closed 2019-06-14 by JWhitleyAStuff

Oyssster, 2020-03-16:
May I ask if you are in a simulation environment or the real velodyne lidar?

JWhitleyWork, 2020-03-16:
@Oyssster This really only applies to using the real lidar because the RP varies somewhat.

nshahrestani, 2020-12-16:
@JWhitleyWork Does the "cut_at_specified_angle" feature fix this? If so, would you mind describing how to set it?

JWhitleyWork, 2020-12-16:
@nshahrestani Please do not bump old threads.

In this case, I do not believe the `cut_at_specified_angle` fixes the problem.

mantkiew, 2023-12-22:
I have just encountered another problem with VLS-128 whereby the sensor is outputting a smaller number of packets per revolution even despite having a 0-359 FOV set. There also seems to be a discrepancy between the number used in the driver and the manual:

[Driver](https://github.com/ros-drivers/velodyne/blob/master/velodyne_driver/src/driver/driver.cc#L67C5-L67C26): 
``` 
packet_rate = 6253.9;
```
Whereas the manual specifies on page 52-53:

> Single return mode: 1 packet/0.1658 ms ≈ 6030.5 packets/second.
> Total packet rate is approximately 6030.5 data packets/second + 94 position packets/second = 6124.5 packets/second,
approximately. Remember, this is an average.

So, is the 6253.9 including both data and position or is it only for data packets? Also, where did the value come from?

The issue #33 applies here as I was operating the lidar in low temperature and despite maintaining the set RPM (I tested with 300, 600, 660, and 1200), the number of packets was only 5700.

So, it seems that (as others have also mentioned), these packets rates are not really that constant and they vary widely depending on the outside temperature. I think that we need to have a more dynamic algorithm that determines the actual packet rate based on the azimuth of the incoming packets - we can simply count them until they complete the full revolution.