ros-drivers/velodyne #246: can not read and show a complete revolution with launch files in velodyne_pointcloud package

wangguojun2018, 2019-06-05:
Dear:
when I use the launch file in velodyne_pointcloud to read pcap files, I can not read a complete revolution. The results in rviz is shown as following:

[image]
when I write a ros node to save pointcloud to pcd file, it is shown as following:
[image]
the following is my launch file
[文字文稿1.pdf](https://github.com/ros-drivers/velodyne/files/3256294/1.pdf)

can you tell me  how to solve it, thank you very much

JWhitleyAStuff, 2019-06-06:
@wangguojun2018 - Have you tried running this with the default configuration (yml) file that is provided for the 64? See if you get the same result.

JWhitleyAStuff, 2019-06-06:
Also, please provide the following information:

**Please complete the following information:**
 - OS and Version: [e.g. Ubuntu 16.04]
 - ROS Version: [e.g. Kinetic]
 - Built from Source or Downloaded from Official Repository:
 - Version: [if from repository, give version from `sudo dpkg -s ros-$ROS_VERSION-velodyne`, if from source, give commit hash]

JWhitleyWork, 2026-11-16:
[written by AI]

Audit update (2026-11-16): I reviewed the attached launch file: it configures an HDL-64E S2.1 CloudNodelet, with `cut_angle=-0.01` (angle-based scan splitting disabled) and `read_once=true`. The corresponding PCAP is not attached, so I could not verify the incomplete revolution. Please provide a shareable capture plus the exact replay command and expected scan boundary/point-cloud output; this is a static review only.

Source: [audited branch code/documentation](https://github.com/ros-drivers/velodyne/blob/29abd0e1361cb7f5eda451d2b51c35eeca45e0d5/velodyne_driver/src/driver/driver.cc#L110-L148).