ros-drivers/velodyne #247: velodyne_msg not getting built

LazyBuilder, 2019-06-12:
-  Ubuntu 18.04
 - ROS Melodic
 - Built from Github using [Tutorial](http://wiki.ros.org/velodyne/Tutorials/Getting%20Started%20with%20the%20Velodyne%20VLP16)

**Describe the bug**
I have just installed the velodyne_driver.
I am able to create point clouds and visualize it in Rviz. I can also subscribe to velodyne_points and see the data.
But when I try to run the basic velodyne node and subscribe to velodyne_packets topic. I get error of velodyne_msg not being built.
[image]

I even tried to build a fresh catkin workspace and re-install just the velodyne_driver. My catkin_make runs without error.

JWhitleyAStuff, 2019-06-14:
@LazyBuilder - This appears to be a sourcing issue. Are you sourcing the workspace after you build it (e.g. `source /home/dts/catkin_ws_dcs/devel/setup.bash`?

JWhitleyAStuff, 2019-11-26:
Closing due to lack of response.

closed 2019-11-26 by JWhitleyAStuff