ros-drivers/velodyne #248: The number of points in single frame in ros message is larger than what veloview show

zorosmith, 2019-06-13:
**Please complete the following information:**
 - OS and Version:Ubuntu 16.04
 - ROS Version: Kinetic
 - Built from Source
 - Version: commit 698f4f00b41cf2a3d7406de21f2d2edc9d236ac2

**Describe the bug**
I set the angle range of VLP16 to [59, 300), 240°. 600RPM.
It means that the point number of each point cloud frame must be less than 19200(16*1800*240/360 = 19200) after removing NAN points.

In VeloView, I can see the point number of each point cloud frame is indeed less than 19200.
These are results: 18422, 18224, 17999, 18764, ...

However, when I launch VLP16 in ros. 
`roslaunch velodyne_pointcloud VLP16_points.launch
 calib:="/home/gordon/ros_ws/src/velodyne/velodyne_pointcloud/VLP16.yaml"`  .
I run the command   `rostopic echo -p /velodyne_points`    to see the point numbers.
I find that point numbers are always larger than 19200 very much.
These are results: 27969, 27618, 27955, 27631, 27944, 27983, ...

I run command   `rosrun pcl_ros pointcloud_to_pcd /input:=/velodyne_points`   and check the *pcd file. There are no NAN points in these files. 

**Expected behavior**
I think that the point number of each point cloud frame in ros is wrong.
Could u provide me some help!

JWhitleyAStuff, 2019-06-13:
@zorosmith - Thank you for the very complete issue request! So, the likely reason that you're seeing additional points is that the driver always assumes a specific number of points per revolution, regardless of what you set on the sensor. To correct this, you can use a fairly new feature: `cut_angle`. Change your launch call to the following:

    roslaunch velodyne_pointcloud VLP16_points.launch cut_angle:=5.24 calibration:="/home/gordon/ros_ws/src/velodyne/velodyne_pointcloud/VLP16.yaml"

This will make sure that each published point cloud always ends at 5.24 radians (300 degrees) azimuth. Let me know if this works for you.

zorosmith, 2019-06-17:
Hello @JWhitleyAStuff!
Thank you for your help! It works for me!

closed 2019-06-17 by JWhitleyAStuff