ros-drivers/velodyne #249: frame_id ignored and published with constant "velodyne"

omrirz, 2019-06-20:
**Please complete the following information:**
 - OS and Version: Ubuntu 16.04
 - ROS Version: Kinetic
 - Built from Source
 - Version: 16bc0c3

**Describe the bug**
velodyne/velodyne_pointcloud/src/conversions/convert.cc sets:
config_.target_frame = config_.fixed_frame = "velodyne";
and thus the frame_id value passed to launch files is being ignored.

**To Reproduce**
Steps to reproduce the behavior:
1.run roslaunch velodyne_pointcloud 32e_points.launch frame_id:="some_name"
2. run rostopic echo -p /velodyne_points/header
3. See fame_id is "velodyne"

**Expected behavior**
fame_id should be the argument passed to the launch file

JWhitleyAStuff, 2019-06-20:
Not sure how I missed this. Thanks for pointing it out! Any chance you could make a PR?

omrirz, 2019-06-23:
Yes. Will try to find time soon.

JWhitleyAStuff, 2019-06-25:
@omrirz - Can you please pull the branch `fix/fixed_frame_target_frame` and verify that this solves your issue? Thanks.

JWhitleyAStuff, 2019-07-01:
Solved by #253.

bmegli, 2019-07-04:
@JWhitleyAStuff , @omrirz 

This is still not working:
- in current master 96b39056bc16c5d3be9d32a709d4cd824189ef40
-in 4f042d92bccb9ba2c01356a3f8f586e8d3208c18 were it was supposed to be fixed 

It is working in 5e02d34cf8fe178844cb1d764a1cbd543596512f which I use

JWhitleyAStuff, 2019-11-20:
@bmegli - Is this still an issue with the current `master`? I haven't been able to look into it recently.

lesterlo, 2019-12-17:
@bmegli 

Yes, even on master branch (3d96504). The frame_id is still ignored.

The target_frame and fixed_frame are "velodyne" even the frame_id on the .launch file is modifed to other name.

pankit7, 2020-01-03:
> @bmegli - Is this still an issue with the current `master`? I haven't been able to look into it recently.

Yes, this is still an issue. As stated by Bartosz, it works in  [5e02d34](https://github.com/ros-drivers/velodyne/commit/5e02d34cf8fe178844cb1d764a1cbd543596512f) but not in the current master branch

JWhitleyAStuff, 2020-01-03:
@bmegli / @lesterlo / @pankit7 - Please try the `fix/convert_frames` branch on this repo. There were a lot of start-up parameters that were not being passed down to the Convert class.

JWhitleyWork, 2020-02-17:
ping @bmegli / @lesterlo / @pankit7 .

JWhitleyWork, 2020-02-17:
Created #330 to address this issue.

lesterlo, 2020-02-18:
@JWhitleyWork 

Due to the COVID-19 in Hong Kong, I cannot go to my lab to test your fix.

I will test it later on.

JWhitleyWork, 2020-02-18:
@lesterlo - thanks for the update. Do not go out and get yourself sick. Be safe!

pankit7, 2020-03-02:
@JWhitleyWork Sorry for the delayed response. You were right about the start-up parameters not being passed down. The fix worked well for me. Thanks!

JWhitleyWork, 2020-03-02:
Thanks, @pankit7!

JWhitleyWork, 2020-03-20:
Solved by #330.

closed 2020-03-20 by JWhitleyWork