ros-drivers/velodyne #252: ROS 2 Port improvements

clalancette, 2019-06-25:
After the ROS 2 Dashing port in https://github.com/ros-drivers/velodyne/pull/251 is merged, there are several followups that we'd like to do.  We want to split them out of that initial port so that they are easier to track.  This issue is a list of the improvements we want to make going forward:

- [ ] Add the ability to change parameters on the fly.  Right now, you can set different parameters during startup.  Changing them after startup (via `ros2 param set`, for instance) appears to work, but doesn't actually change the values used at runtime.  We should implement the `param_change_callback` so that we can change configuration on the fly.
- [ ] Make the nodes composable.  Right now, all of the nodes are separate, meaning a lot of data is being passed back and forth.  If we made them composable, they could share pointers, greatly improving their performance.  We're currently waiting on ros2/rclcpp#758 to land and to make it into Dashing before making them composable.
- [ ] Port the integration tests.  Currently the integration tests are disabled, but it would be great to re-enable them and make sure they pass in ROS 2.
- [ ] Re-enable parts of logging.  The current port disables some of the logging because it wasn't clear the best way to pass the necessary logging pointer to where it is needed.  This should be reconsidered and passed, then we can re-enable the logging.
- [ ] Get diagnostics working.  In the current ROS 2 port, all of the diagnostics code is disabled.  This is mostly because diagnostics is not yet ported/available in ROS 2 Dashing.  Once it is available, we should re-enable the diagnostics stuff and use it.
- [ ] Make "cut_at_specified_angle" the default with an angle of 2*pi radians.  The current version defaults to a specific number of packets per revolution (npackets). Since the rotation rate of the sensor can vary, this causes over- and under-scanning in the generated point clouds. I think we should just do away with the npackets value and calculate some max estimated value for possible number of packets for pre-allocation purposes, then use the cut_at_specified_angle feature always.
- [ ] Set "organize_point_cloud" to true by default
- [ ] Rename the node names and file names to be consistent.

JWhitleyAStuff, 2019-06-25:
@clalancette - Would you mind creating separate issues for these so we can track them/discuss them independently? I've created a Dashing milestone that we can attach them to for tracking.

clalancette, 2019-06-26:
> @clalancette - Would you mind creating separate issues for these so we can track them/discuss them independently? I've created a Dashing milestone that we can attach them to for tracking.

Will do.

clalancette, 2019-06-26:
OK, split into #257, #258, #259, #260, #261, #262, #263, and #264.  Closing out this larger issue.

closed 2019-06-26 by clalancette