ros-drivers/velodyne #264: ROS 2 Dashing improvement: Rename the node names and file names to be consistent

clalancette, 2019-06-26:
RIght now, the node names aren't consistent (some have a "_node" suffix and some don't, some have a "velodyne_" prefix and some don't), and the file names don't necessarily match the node names.  Change all of this to be more consistent.

JWhitleyAStuff, 2019-11-03:
@clalancette - Would you consider this to be complete by #254?

clalancette, 2019-11-05:
> @clalancette - Would you consider this to be complete by #254?

Not quite.  I renamed the node names as seen by ROS, but I totally forgot to rename the binaries themselves.  So I'd leave this open for now (I may have some time to complete this later this week).

clalancette, 2019-11-07:
Actually, never mind, I think this is good now, so I'll close this out.  Thanks!

PR #254 (2019-07-01): Dashing: Rename nodes.
As discussed in https://github.com/ros-drivers/velodyne/pull/251,
we should make the default node names a lot more consistent.
Here, rename the nodes to all be
velodyne_{driver,convert,transform,laserscan}_node.  I also
renamed the "cloud_node.cc" file to be "convert_node.cc", so it
matches the rest of the code.  Finally, I removed the unused
nodelets.xml file from velodyne_pointcloud.

@JWhitleyAStuff @klintan FYI

Fixes #264
closed 2019-11-07 by clalancette