ros-drivers/velodyne #267: A possible breaking change has been introduced in the TransformNodelet

RonaldEnsing, 2019-07-01:
I use the TransformNodelet to compensate for ego-motion. This gives me a more accurate point cloud in a global (odom or map) frame. All of this works fine when I use velodyne version 1.5.2.

However, when I am using master branch (https://github.com/ros-drivers/velodyne/commit/458bc7c72f607e7e99a93f5ad2ce79cc79b40fc1), the point clouds no longer seem to be corrected on the individual scan packet basis. It is as if the TransformNodelet does not transform individual scan packets, but only the entire scan message.

In version 1.5.2 it is clear that individual packets are transformed using the time stamp (https://github.com/ros-drivers/velodyne/blob/1.5.2/velodyne_pointcloud/src/conversions/transform.cc#L100). In master branch it seems that there is no such correction anymore (https://github.com/ros-drivers/velodyne/blob/master/velodyne_pointcloud/src/conversions/transform.cc#L134), which conforms to my experimental findings that the point clouds look less accurate.

JWhitleyAStuff, 2019-07-01:
@RonaldEnsing - I think you're correct. From the history, it looks like this is something @spuetz removed when he was removing the PCL dependency. Thoughts on this, @spuetz?

JWhitleyAStuff, 2019-07-01:
In the mean time, @RonaldEnsing, would you like to submit a PR to add this code back?

spuetz, 2019-07-01:
Hi guys, I'll check that tomorrow morning. Shouldn't be a big thing to fix.

spuetz, 2019-07-01:
I had a quick look into the code. I'll fix that tomorrow morning with a few lines and open a PR.

spuetz, 2019-07-02:
@RonaldEnsing Could you try #269?

RonaldEnsing, 2019-07-02:
> @RonaldEnsing Could you try #269?

I have tried #269 and it looks all good again.

PR #269 (2019-07-02): fix for #267, transform each packet
This should fix #267.
closed 2019-07-02 by JWhitleyAStuff