ros-drivers/velodyne #275: Packet containing angle overflow

crankler, 2019-07-12:
I use ubuntu 16.04 and ros_kinetic
when I use this master package and I found there can ger scan directly,
but when  I launch VLP16_points.launch, I get this warning:

> Packet containing angle overflow, first angle: 35962 second angle: 0

also I add  this topic in rviz, and there have warning as follow:

> [ WARN] [1562916979.098222693]: VelodyneLaserScan: PointCloud2 fields in unexpected order. Using slower generic method.

So can someone give me advice ? Thank you!

kaalessi, 2019-08-21:
I am also running into this warning, except on ubuntu 18.04 and ros_melodic

JWhitleyAStuff, 2019-08-22:
@crankler / @kaalessi - Try enabling the `cut_angle` feature by setting it to 2*pi (6.28318530718). Let me know if this fixes the problem.

kaalessi, 2019-08-26:
@JWhitleyAStuff Thank you for your response. I have changed the cut_angle in the launch file to "2*pi" and am still running into the warning "Packet containing angle overflow, first angle: 35979 second angle: 17". What was the reason you recommended 2*pi? Is there something else I could change to eliminate this warning? Thanks

doldd, 2019-09-03:
I think the problem is in line 352 `	if(raw_azimuth_diff < 0)//raw->blocks[block+1].rotation - raw->blocks[block].rotation < 0)` from file rawdata.cc. In my opinion the if statement should check azimuth_diff instead to cover the angle overflow.

he-guo, 2019-10-23:
i also have this problem,when I add this topic in rviz,  this warning is only available at this time,have you solved it?thinks.
[image]

kaalessi, 2019-10-24:
@he-guo No, I have not yet been able to solve this problem. I'm getting the same thing  as you when rviz is open. changing the cut_angle did not seem to eliminate this error.

JWhitleyAStuff, 2019-11-20:
This error comes from the `velodyne_laserscan` package, which was created by @kmhallen. This may be due to changes that were made in https://github.com/ros-drivers/velodyne/pull/214 which remove PCL as a dependency but also change the order of the objects in the point cloud, which may cause a problem for this package. @kmhallen - Would you mind looking into this?

kmhallen, 2019-11-20:
I can look into it. If the `PointCloud2` structure changed, `velodyne_laserscan` will take the more generic code path. It should be a simple fix to add a case for the new structure.

VorpalBlade, 2019-11-21:
@kmhallen:
> I can look into it. If the `PointCloud2` structure changed, `velodyne_laserscan` will take the more generic code path. It should be a simple fix to add a case for the new structure.

That does indeed seem to be the case, see bug #292.

kmhallen, 2019-11-21:
See PR #316

JWhitleyAStuff, 2019-11-22:
@crankler @doldd @kaalessi @VorpalBlade @he-guo - #316 has now been merged. Can you please test this from `master`?

VorpalBlade, 2019-11-25:
@JWhitleyAStuff 
I'm using a VLP-16 and I'm still getting the "Packet containing angle overflow" message. However, the "VelodyneLaserScan: PointCloud2 fields in unexpected order. Using slower generic method." message mentioned in bug #292 does seems fixed now. Did you by any chance mix up the bug IDs?

JWhitleyAStuff, 2019-11-25:
@VorpalBlade - @crankler originally mentioned both issues. Thanks for testing #292 and reporting that it worked for you. I will leave this open for the remaining error.

ryanmeasel, 2020-02-28:
We're running two VLP16s concurrently and we receive the "Packet containing angle overflow" warning on each unit every 60 seconds. It's periodic. And because we launch the nodes at the same time, the warnings happen together as well.

ajwiley, 2020-08-25:
Has anyone found a solution to this yet? I'm only running one VLP16 on a Clearpath Husky, but I'm having the exact same thing as @ryanmeasel happen to me.
I run VLP16_points.launch, I open up RViz, and then I start seeing "Packet containing angle overflow, first angle: [big_number] second angle: [small_number]" repeating every 60 seconds.

kaalessi, 2020-08-25:
@ajwiley I was working with a VLP16 yesterday and am still receiving an angle overflow warning when I have Rviz open or am recording with rosbag. Earlier this summer, I updated to the most recent package, and am still receiving the warning. Unfortunately, due to COVID-19, I have had limited on-site time working with the lidar unit.

Timple, 2021-01-22:
Can anybody already verify that #285 would fix this?

DarrenWong, 2021-05-14:
I have a similar issue with VLP 16 (HDL 32 is ok)  as @ryanmeasel @ajwiley .with version [1.6.1](https://github.com/ros-drivers/velodyne/tree/1.6.1). As the ROS_WARN is set to print every 60 s, such that we can consider the condition of warning is always true. 
Any suggestion? Seems it might drop a few frames if the second condition not match again. I am using Melodic under Ubuntu 18.04 Bionic.

https://github.com/ros-drivers/velodyne/blob/4462666aef946dc87907e7ef4f9f083a04b31b55/velodyne_pointcloud/src/lib/rawdata.cc#L672-L683

[image]

kennedyaidan, 2021-06-30:
I too am experiencing a 60s periodic "Packet containing angle overflow" warning when running my VLP16. As @DarrenWong points out, it's likely always true and only prints out at that frequency. Any suggestions on how to fix, and/or input on what kind of issues this may have would be appreciated.

DarrenWong, 2021-07-01:
@kennedyaidan  I just ignore it currently as it still can working on the SLAM method~... But my experiments show that it will skip a few frames (not consecutive ) compare to HDL-32 if collect together.

happengor, 2021-11-03:
> Can anybody already verify that #285 would fix this?

I have tested it, #285 doenst fix it, but it makes the motion distortion predictable and the motion distortion now looks  reasonable, which in my opinion is a great modification.

GianlucaCerilli, 2022-01-17:
I am also using a Velodyne (Puck VLP-16) on a robot (it is connected to a Mini PC Intel NUC with Ubuntu 20.04 and ROS Noetic) and I get the same warning "_Packet containing angle overflow_".
The problem is that, when I get this warning the robot also does some weird movements and this can be dangerous.

Did anyone find a solution for this problem? Thanks in advance

GianlucaCerilli, 2022-01-18:
> > Can anybody already verify that #285 would fix this?
> 
> I have tested it, #285 doenst fix it, but it makes the motion distortion predictable and the motion distortion now looks reasonable, which in my opinion is a great modification.

Hi @happengor ,
how did you test the code at [#285](https://github.com/ros-drivers/velodyne/pull/285)? Because there are some files/functions that no longer exist in master or that are changed.

happengor, 2022-01-18:
> Hi @happengor , how did you test the code at [#285](https://github.com/ros-drivers/velodyne/pull/285)? Because there are some files/functions that no longer exist in master or that are changed.

hi, i have adjusted the new code base according to https://github.com/ros-drivers/velodyne/pull/285/files . But first u need understand the code structure. U could also git clone this https://github.com/MapIV/velodyne/tree/feature/precise_scan_separation and test it.

GianlucaCerilli, 2022-01-18:
> hi, i have adjusted the new code base according to https://github.com/ros-drivers/velodyne/pull/285/files . But first u need understand the code structure. U could also git clone this [...]

There is currently a build error on the "frame_id" when building https://github.com/MapIV/velodyne/tree/feature/precise_scan_separation. 
Do you have the code that you have adjusted according to https://github.com/ros-drivers/velodyne/pull/285/files? For example, the conversion files do not exist anymore.

happengor, 2022-01-18:
> There is currently a build error on the "frame_id" when building https://github.com/MapIV/velodyne/tree/feature/precise_scan_separation. Do you have the code that you have adjusted according to [...]

For ur first try, u can hard code it : -->> change config.frame_id to "velodyne"  . It's ur definition of frame_id for transformation of coordinate. 
I think the source code by catkin_make may be redirected to /opt/ros/noetic/include/velodyne_pointcloud/TransformNodeConfig.h.

kennedyaidan, 2022-01-18:
> I am also using a Velodyne (Puck Lite VLP-16) on a robot (it is connected to a Mini PC Intel NUC with Ubuntu 20.04 and ROS Noetic) and I get the same warning "_Packet containing angle overflow_".
> 
> [image]
> 
> The problem is that, when I get this warning the robot also does some weird movements and this can be dangerous.
> 
> Did anyone find a solution for this problem? Thanks in advance

I'm commenting just to put into question that the "Packet containing angle overflow" is causing any sort of weird movement or dangerous behavior. The overflow error appears to be continuous, and only prints out every 60s or so. I can't see how the printout frequency of a warning could have any effect on the actual behavior of the robot. I'd suggesting looking elsewhere to find the root cause of that "weird" behavior. If you have more data suggesting it is actually causing "weird"' behavior please let me know!

GianlucaCerilli, 2022-01-19:
> > There is currently a build error on the "frame_id" when building https://github.com/MapIV/velodyne/tree/feature/precise_scan_separation. Do you have the code that you have adjusted according to https://github.com/ros-drivers/velodyne/pull/285/files? For example, the conversion files do not exist anymore.
> 
> For ur first try, u can hard code it : -->> change config.frame_id to "velodyne" . It's ur definition of frame_id for coordinate transformation . I think the source code may be redirected to /opt/ros/noetic/include/velodyne_pointcloud/TransformNodeConfig.h during catkin_make .

Thank you very much @happengor ,
I have tested the [precise_scan_separation](https://github.com/MapIV/velodyne/tree/feature/precise_scan_separation) code but unfortunately I still have the warning. I also could not see any relevant change...

**ros_drivers/Velodyne**
[image]

**precise_scan_separation**
[image]

Anyway, I tested the [ros_drivers/Velodyne](https://github.com/ros-drivers/velodyne.git) on both Kinetic (Ubuntu 16) and Noetic (Ubuntu 20) and I have this warning only on Noetic. Maybe they wrote something to fix this warning for Kinetic in the two years from the opening of this issue.

@kennedyaidan 
I agree, this is weird and we are investigating more. Maybe there could be some communication problem...I don't know. 
I'll write an update If I find something interesting.

ahmetsaglam, 2023-03-08:
any updates on how to fix this warning?

tersite1, 2024-02-20:
I had an same issue, Ubunto 20.04 , ROS noetic 
but I just solved by simply reinstall ROS, 
try
`sudo aot install ros-noetic-desktop full` 

than delete catkin_ws

`rm catkin_ws`

```
mkdir catkin_ws
mkdir src

```

than build catkin again 

```
sudo apt-get install ros-noetic-velodyne
cd catkin_ws/src
git clone https://github.com.ros-drivers/velodyne
cd ..
catkin_make
cd src

roslaunchvelodyne_pointcloud VLP16_points.launch
rosrun rviz rviz -f velodyne
```

hope this can be helpful thx

havetosaynice, 2025-04-19:
is it working？

JWhitleyWork, 2026-11-16:
[written by AI]

Audit update (2026-11-16): The separate PointCloud2 field-order warning was addressed by [#316](https://github.com/ros-drivers/velodyne/pull/316), but the angle-overflow behavior remains in ROS1 `master`: a negative raw azimuth delta uses the previous delta or skips that block if no previous delta exists. The current ROS2 code retains the fallback/skip path, with the warning itself commented out. [#285](https://github.com/ros-drivers/velodyne/pull/285) remains open, and a commenter reports it did not remove this warning. No hardware or capture replay was performed.

Source: [audited branch code/documentation](https://github.com/ros-drivers/velodyne/blob/29abd0e1361cb7f5eda451d2b51c35eeca45e0d5/velodyne_pointcloud/src/lib/rawdata.cc#L648-L710).