ros-drivers/velodyne #277: VLP16 Timestamp not correct with attached GPS

alec-gurman, 2019-07-29:
I am working with a Velodyne VLP16 and have a GPS mouse attached so that i can perform accurate timestamping for incoming packets. The Velodyne web interface show's that the gps is connected and reading a GPS position, however the PPS (regardless of outdoors) is always absent.

I have enabled the `use_gps` flag in the launch file, my understanding is, the timestamp is at location 1200 of the position packet, and this is a microsecond counter since TOH. The PPS signal is used for subsecond accuracy whilst the GPRMC component of the nmea scentene is used for the minutes and seconds, which is then converted to TOH and used to update the velodyne's internal counter with proper UTC time. 

The main problem is the reported ROS timestamp for the packets is way off compared to the system clock (or even to correct UTC time) something like 25 minutes ahead. Is this because there is not a valid PPS signal and therefore the TOH is not updated as it cannot detect a rising edge of the PPS? Is PPS required for time synchroziation to occur, or should just the nmea scentence be enough for minutes and seconds (if i dont care about subsecond accuracy)?

My understanding is that the current reported timestamp is way off as time synchronization is not occuring and therefore the timestamp is simply obtained from the on board oscillator and hence the counter is incremented starting from the time the velodyne was first turned on.

If anyone is able to confirm these points for me that would be great, particualy if the PPS is **required** in order to perform time synchronization. 

Cheers.

JWhitleyAStuff, 2019-08-28:
@alec-gurman - I'm sorry for the long delay. I actually do not know how to answer this - it is something you would have to ask Velodyne support directly. Sorry I can't be of more help.

closed 2019-08-28 by JWhitleyAStuff

arukan404, 2022-11-29:
We are observing this too right now with the ros2 nodes

JWhitleyWork, 2022-12-04:
@arukan404 Please do not bump closed issues. Create a new one if you are having a problem and link to the closed one, if necessary.