ros-drivers/velodyne #281: Add a per-point timestamping of data

kaarta-SHanna, 2019-08-20:
**Is your feature request related to a problem? Please describe.**
The driver does not output the timing information for a pointcloud, so you cannot precisely transform a point into 3d space while moving

**Describe the solution you'd like**
Output the timestamp of each point when processing the pointcloud

**Describe alternatives you've considered**
You can use tf, but this could be more useful when running a custom unwarping function where you need point time information

**Additional context**
Implemented here:
https://github.com/kaarta/velodyne/tree/feature/addPerPointTiming

kaarta-SHanna, 2019-08-20:
I have not yet put in the code for any 64 line scanners, but the vlp32, hdl32 and vlp16 should work. I also need some feedback on the optimal way to get the velodyne model from the parameter server (cloud vs driver private namespaces aren't easy to access). Right now it's hard coded to look for the param in: the namespace "/velodyne_nodelet_manager_driver"

plitkey, 2019-09-27:
I tried to time stamp VLP16 data and got messages with extra time field that is always zero.
 
I started xterm and gave command
rosrun velodyne_pointcloud cloud_node _calibration:=/home/me/catk/src/velodyne/velodyne_pointcloud/params/VLP16db.yaml

started recording a bag in another xterm, recording topic /velodyne_points 

in yet another xterm started playing a bag that has topics /VLP16/velodyne_packets and /VLP16/velodyne_points (the /VLP16/velodyne_packets are remapped to /velodyne_packets)

The rqt-graph looks ok, new bag is written with point_cloud2 messages. The new messages are exactly same as the original ones, only extra field with zeros as time is added. I was expecting the points to slightly move based on what is done in rawdata.cc (line 380 and 388) and of course to get the time for each point.

How can I get per point timing?

kaarta-SHanna, 2019-09-30:
@plitkey I will look into this as soon as I get back in town this week. We need to pass the model parameter through the launch file better so I haven't finished that part yet.

plitkey, 2019-10-01:
Great @kaarta-SHanna ! I already tried with a rude stabbing that forced the rawdata.cc to use VPL16 model, I got rid of the warning, but unfortunately could not find a way to get results. Somehow the timing from the bag was not what the node expects. I tried with different --clock and -use-sim-time options and tried starting the bag player before the point_cloud node but could not find a way to produce the point times.

kaarta-SHanna, 2019-10-02:
@plitkey I just modified the launch files on our fork to support the "model" parameter. so it should work now if you run the driver with normal launch options. I just tested with a VLP16 bag file, and a hdl32 pcap file. It would be good if someone else could test this too.

Here I have an image of rviz showing the vlp16 "time" field for a full revolution (which is approximately 0.1 seconds for a 600RPM vlp16)

[image]

plitkey, 2019-10-03:
There seems to be a header missing, I could not build. 
src/velodyne/velodyne_driver/include/velodyne_driver/input.h:58:18: fatal error: pcap.h: No such file or directory
 #include <pcap.h>

plitkey, 2019-10-03:
Thanks!... I could not get it working though. Not with the previous rosrun command with model added (_model:=VLP16) or using  a launch file. I could get the node running with a launch file, but the messages are not going from bag to pointcloud node and from there to record bag as they should. An example of a launch file with a bag would be appreciated.

Meanwhile, I have timestamped points from the rude hack I made previously and am wondering about this line:
 float time_diff_start_to_this_packet = (pkt.stamp - scan_start_time).toSec();

Why difference?
I think I would need pkt.stamp of it is the time that is also in imu packets. Now my times are - something... And if I rudely hack and use (pkt.stamp).toSec() all time stamps are the same. (Is this from using float?)

kaarta-SHanna, 2019-10-03:
@plitkey Here's an example command for running an HDL-32e pcap file:
TERMINAL 1: roslaunch velodyne_pointcloud 32e_points.launch pcap:=$HOME/recordings/hdl32e_pcap/pcap.pcap

Here's an example for running from a bag file:
TERMINAL 1: roslaunch velodyne_pointcloud VLP16_points.launch
TERMINAL 2: rosbag play velo.bag

As for the using a difference: we are trying to show the timestamp of each point relative to the header stamp of the pointcloud. Thus, if we are timestamping the pointcloud with the timestamp of the last packet in the pointcloud, we can have negative numbers. If we timestamp the pointcloud using the first packet's timestamp (which is a configuration option added by #280), you will have positive seconds since the timestamp. To get the absolute timestamp for any given point, add the point's time field to the pointcloud's header timestamp.

Since each packet is processed individually, I use time_diff_start_to_this_packet to tell me the diff from the first packet's time to the current packet's time (so, packet 70 in a given scan will have an offset to the first packet in the pointcloud)

plitkey, 2019-10-04:
Thanks! This was very helpful for me.
(I learned that the bag file must be started before the launch file to have points processed, otherwise there is no connection between the bag and the point_cloud node)

plitkey, 2019-10-04:
Playing with launch files... I do not understand why making a launch file that plays and records rosbag and includes VLP16_points.launch starts OK but then gives 
[ WARN] [1570182432.527729833, 1561980147.159589384]: Velodyne poll() timeout
[ERROR] [1570182432.527857239, 1561980147.159589384]: DriverNodelet::devicePoll - Failed to poll device.
Driver essentially wants either pcap defined or it is polling the port - right? How does bag fit into this?

Driver is not needed in what I want to do so it is surprising that VLP16_points.launch works with bag file when started in terminal.

If I modify the VLP16_points.launch file so that the driver is not started but only point_cloud node, the messages do not connect the bag and the point_cloud node.

kaarta-SHanna, 2019-10-04:
This is a separate issue not related to this feature

TixiaoShan, 2019-11-01:
@kaarta-SHanna 
Just want to say thank you for implementing this function in the driver.

johnsongash, 2019-11-13:
@kaarta-SHanna 
Not quite understand how to get every point's timestamp from the implement,
any explanation will be thankful!

kaarta-SHanna, 2019-11-13:
The final output pointcloud has a field called time which is the time of an individual point's firing relative to the pointcloud's header timestamp.

kaarta-SHanna, 2019-11-18:
Looks like this was closed in PR #290

closed 2019-11-18 by kaarta-SHanna

jaeseok4104, 2023-06-06:
how can i use this feature?

JWhitleyWork, 2023-06-07:
@jaeseok4104 Please stop replying to dead threads like this. If you have a question, create a new one.