ros-drivers/velodyne #282: how to get vertical beam angle for each rings of velodyne lidar

ajay1606, 2019-08-21:
Hello all, 

I am using a Velodyne 64 SE model lidar sensor, for some algorithm implementation, am looking for the vertical angle of each ring. 

Although I can get that from the 3D coordinates, because of some obstacles in the environment, its very difficult to get consistent verticle angle of the laser beam.

Is there any way that I can get actual verticle beam angle in the Velodyne 64 sensor.

Your response greatly appreciated.

Regards,
Ajay

JWhitleyAStuff, 2019-08-21:
@ajay1606 - The lasers are evenly spaced between -26.9 degrees and +2.0 degrees (approximately 0.4203125 degrees apart). You should be able to calculate the angle of each beam using these data. There are very minor offsets for each sensor but they are provided in the .yaml configuration file that Velodyne provides with the sensor. Please let me know if you have any other questions. Thanks.

JWhitleyAStuff, 2019-08-21:
Correction: I guess it depends on exactly which model you have. Here are the specifications for the S2, which is slightly different than the SE: https://velodynelidar.com/lidar/products/brochure/HDL-64E%20S2%20datasheet_2010_lowres.pdf

closed 2019-08-21 by JWhitleyAStuff

ajay1606, 2019-08-22:
@JWhitleyAStuff Thank you so much for quick response as always, So **(-24.8 deg)** corresponds to the **ring 0**  and t+2 deg corresponds to the ring 63 in the **<velodyne_pointcloud::PointsXYZIR>** .

Thanks