ros-drivers/velodyne #287: Question: PointXYZIR definition was deleted: why?

beetleskin, 2019-09-02:
The point definition was valid for all Velodyne Pointclouds. Workin with pcl requires to define such a point, so why removing it? How I have to guess the point type in order to work with the driver's pointcloud and pcl.

JWhitleyAStuff, 2019-09-03:
Removing the PCL dependency significantly increased the speed of the node. In addition, it allowed us to break away from the PCL definitions of what point clouds should contain and allows for adding more containers for point clouds (such as the OrganizedCloudXYZIR). The result is still published as a standard ROS [sensor_msgs/PointCloud2](http://docs.ros.org/melodic/api/sensor_msgs/html/msg/PointCloud2.html) which contains enough information for the cloud to be converted to a PCL cloud using [fromROSMsg](http://docs.ros.org/hydro/api/pcl_conversions/html/namespacepcl.html#af662c7d46db4cf6f7cfdc2aaf4439760). Where are you having a problem with the container? What are you trying to do with it and from where?

beetleskin, 2019-09-05:
As stated, fromROSMsg creates a `PointCloud<T>`. One needs the T. If you get the T wrong (wrong packing, order, atomic types) the conversion does not work without further notice (the cloud is just scrambled) afaik. 
So, with removing the PointXYZIR, every user of this driver working with PCL will have to implement PointXYZIR on his own. I found it very convenient and straight-forward to use. Also, there is simply no other choice for the point type other than PointXYZIR, right? So why not providing it.

Now users will be puzzled on how to convert the  message correctly. As for me, I just copied the old definition from the driver over to my code.

beetleskin, 2019-09-05:
Despite that, a lot of users build against this type right now (provided by the package released to kintetic/melodic atm). Removing it is a big ABI breaking change. Everyone using it needs to adapt. I'm not saying one should not do that, but most ros packages avoid that and I wonder what the reason was.

Or is there another way?

JWhitleyAStuff, 2019-09-05:
@spuetz - It looks like you were the implementer in #214. Do you have any feedback on this?

spuetz, 2019-09-06:
Hi @beetleskin Since PCL is not maintained anymore, the conversion from PCL to the `sensor_msgs/PointCloud2` is not necessary and takes a lot of time and PCL is a quite large dependency it is better to not having it anymore. Also with PCL you need to know at compile time which fields you want to have in the cloud. 

I suggest: The cleanest solution for you would be to add a `velodyne_pcl` package having the PCL dependency and providing the PCL point type structs. That should be enough, right? You could implement that package and open a PR. @JWhitleyAStuff, what do you think?

JWhitleyAStuff, 2019-09-11:
@spuetz - I would tend to agree. However, if you have time, you might be a much better person to implement this - especially since you removed the dependency originally. I'll let you two discuss that.

beetleskin, 2019-09-23:
Well for now I'm more or less happy with copying your original PointXYZIR definition into my project. If no one else stumbles upon this, I guess there's no need to act. Maybe I'm the last user of PCL ;) If not, it might be nice to re-add this definition for the reasons mentioned above.

Mostly I was curious if I missed something out here.

----

> Also with PCL you need to know at compile time which fields you want to have in the cloud.
Since the Hardware does not change, the fields won't change as well, right? Since you know your Velodyne at compile time, you know the Point definition?

spuetz, 2019-09-23:
> Since the Hardware does not change, the fields won't change as well, right? Since you know your Velodyne at compile time, you know the Point definition?

Sure you could provide all field all the time with respect to the selected container. However, sometimes you don't need all field and it could be beneficial for the network usage / capacity to minimise useless sent data. 

I'll add a `velodyne_pcl` package with this pcl point types when I've time.

beetleskin, 2019-09-24:
> Sure you could provide all field all the time with respect to the selected container. However, sometimes you don't need all field and it could be beneficial for the network usage / capacity to minimise useless sent data.

Right, I didn't think of that.
How would I configure _velodyne_pointcloud_ to, lets say, only publish _X,Y,Z_ and _I_ (leave out _R_)?

I guess http://wiki.ros.org/velodyne_pointcloud?distro=melodic#C.2B-.2B-_API is outdated either way?

> I'll add a velodyne_pcl package with this pcl point types when I've time.

Thanks a lot!

Zhang-Qinghui, 2019-10-30:
Thank you very much.I got it.

Zhang-Qinghui, 2019-10-31:
Hi,spuetz.I want to convert the point cloud from the VLP16 to depth image, but when I run "rosrun pcl_ros convert_pointcloud_to_image input:=/velodyne_points output:=cloud_image",there is a error "Error in converting cloud to image message: No rgb field!!" . Do you know how to solve this? Thank you very much.

spuetz, 2019-11-01:
There are no color information in the point cloud ;) What do you want to do in general?

Zhang-Qinghui, 2019-11-01:
Thanks for your reply. I want to conver  the point cloud from the VLP16 to depth image, so I can deal with depth image to detect dynamic obstacle in real time.

spuetz, 2019-11-01:
But why not using the organized point cloud for that? Organized clouds could also be accessed in a matrix fashion.

Zhang-Qinghui, 2019-11-01:
I want to use Background subtraction in depth image to detect dynamic obstacle. and my VLP16 is fixed in the scene. In addition, do you have some advise about dynamic obstacle detection. @spuetz

spuetz, 2019-11-01:
You cloud use an octree, e.g, octomap and compute the difference over time. I think that is much simpler.

Zhang-Qinghui, 2019-11-02:
Thank you very much.

Zhang-Qinghui, 2019-11-12:
@spuetz hi,spuetz. you say that organized clouds could also be accessed in a matrix fashion.And I have tried, but I have another question.Does the point under the same index number in the ordered point cloud of different frames come from the same position in the actual space? thank you in advance.

spuetz, 2019-11-12:
@Zhang-Qinghui  Plese open a new Issue for that questions. I think this issue ist the wrong space for these detailed questions.

Zhang-Qinghui, 2019-11-12:
Thanks.Because this problem has some connection with the above, I put him here.Now I will open a new issue.

JWhitleyAStuff, 2020-01-20:
@spuetz - Have you had any time to think about the `velodyne_pcl` node that you suggested?

spuetz, 2020-01-20:
Not yet. Is it urgent?

JWhitleyAStuff, 2020-01-20:
I'd say... a little. Though that's a bit my fault. I haven't released a new version of the driver in almost a year and I'd like to get a new one out with the great improvements that have been made. However, I'm afraid that releasing the current version which produces clouds that are not compatible with PCL is going to cause trouble for lots of people and I'd like to have something to point them to when they ask about it.

YoshuaNava, 2020-02-28:
Hi all,
Is this PCL feature deemed critical for a new release? As @spuetz said, PCL is a big dependency, and there are users that benefit from other point cloud libraries (libpointmtcher, open3d, etc), and could find this extra dependency unnecessary.

YoshuaNava, 2020-03-19:
Dear @JWhitleyWork,
I gather that you are the maintainer of this repo. By when do you estimate that a new release could happen? There are many features and fixes introduced during 2019 that are great, and regular ROS users can not make use of them.

Thank you in advance.

Best regards,
Yoshua Nava

JWhitleyWork, 2020-03-24:
@YoshuaNava I agree, but the problem is that this particular change, introduced by #214, completely breaks the existing API. I don't really know why I hadn't noticed that when I merged it but I didn't and now it's a problem.

@spuetz This has become a critical issue. I need you to do one of the following as soon as possible:

1. Re-introduce the PCL dependency and add the `velodyne_pointcloud/point_types.h` file back;
2. Find a way to create a new `velodyne_pointcloud/point_types.h` file which provides the same functionality as the previous PCL-dependent version; or
3. Tell me that you simply can't do it or can't make time to do it right now so I can fix it.

spuetz, 2020-03-24:
@JWhitleyWork, I will implement a `velodyne_pcl` package with the necessary conversion file. This means that packages which what to use the conversion have to depend on `velodyne_pcl`. 
Is that ok for you?

JWhitleyWork, 2020-03-24:
@spuetz Yes, that is OK. Thank you very much for this. Once that's in and tested, I'll update the ROS wiki page with migration details.

JWhitleyWork, 2020-03-28:
@YoshuaNava Can you please test #335? I no longer have access to the hardware to test this. Once it has been tested, I will merge it and release a new version.

YoshuaNava, 2020-04-02:
@JWhitleyWork Yes, I can test it, but approximately next week.

In order to run the tests in the best way possible, could we create a list of items to test?

JWhitleyWork, 2020-04-02:
> In order to run the tests in the best way possible, could we create a list of items to test?

I will provide some tests but I need @spuetz to address one additional issue before testing. Please hold. Thanks for offering to help!

JWhitleyWork, 2020-04-29:
OK, so @spuetz and @YoshuaNava - I've been thinking about this long and hard. Here's my dilemma: We've already broken the API in `master` by removing the PCL dependency. However, there are plenty of projects out there using Melodic that haven't noticed simply because they're using the released version and I haven't released a new version from `master` in some time. Here's my plan:

1. Go back to the commit before https://github.com/ros-drivers/velodyne/commit/b4ad0ae9b601d1f9fa234a1bbaaa8cbc179e4bcb introduced the breaking changes and create a `melodic` branch from there.
2. Do my best to cherry-pick changes that were made after that point onto the `melodic` branch without breaking the API.
3. Have someone test the `melodic` branch.
4. Release the `melodic` branch into the build farm as a Melodic release.
5. Consider `master` to be targeted at Noetic.

Not only does this remove the need for #335, it keeps from breaking the Melodic API. Noetic will then receive the new, no-PCL API.

Any problems/concerns with this approach?

beetleskin, 2020-04-29:
you might wanna open another issue for that whole topic ;) although my question started this conversation, it seems not too much related to the direction it took

JWhitleyWork, 2020-04-29:
Thanks, @beetleskin. I guess our discussion mostly answered your question. I will create a new issue to discuss.

closed 2020-04-29 by JWhitleyWork

YoshuaNava, 2020-04-29:
@JWhitleyWork Sounds good.