ros-drivers/velodyne #288: Can't run velodyne_pointcloud

Mixiaomiupup, 2019-09-11:
After I run the velodyne_driver node correctly,here is the message:
[ INFO] [1568169846.920503593]: Velodyne VLP-16 rotating at 600 RPM
[ INFO] [1568169846.922195451]: publishing 76 packets per scan
[ INFO] [1568169846.926666203]: Cut at specific angle feature deactivated.
[ INFO] [1568169846.938948504]: Reconfigure Request
[ INFO] [1568169846.941202167]: expected frequency: 9.921 (Hz)
[ INFO] [1568169846.942881787]: Opening UDP socket: port 2368
Then I run cloud_node and transform_node ,both of them  meet the ERROR messasg as follows:
[ERROR] [1568170441.444897651]: No calibration angles specified! Using test values!
I can't handle this problem,please help me!

JWhitleyAStuff, 2019-09-11:
@Mixiaomiupup - Are you able to run the entire stack using `velodyne_pointcloud/launch/VLP16_points.launch`?

JWhitleyAStuff, 2019-09-16:
@Mixiaomiupup - You must specify the calibration YAML file and pass it to the `velodyne_pointcloud` node as the 'calibration' parameter.

closed 2019-09-16 by JWhitleyAStuff