ros-drivers/velodyne #306: Missing some data when replay pcap in VLP16

TheGreatGalaxy, 2019-11-05:
- OS and Version: [Ubuntu 16.04]
 - ROS Version: [ Kinetic]

Hi, I am using VLP16 to fuse the pointscloud with VLP32. But when i using the driver to replay pcap, it seems missing some angle data and the missing data is oritenting when replay pcap. I am sure that the pcap data is right because of verified by VeloView. See as following pictures:

[image]
How should to resovle it?

TheGreatGalaxy, 2019-11-05:
[image]

JWhitleyAStuff, 2019-11-20:
@TheGreatGalaxy - If you record these data using the real sensor and record with a `rosbag`, does replaying the `rosbag` show the same effect or is it only when you replay the `pcap` file using the driver? Have you tried replaying the `pcap` file with something like Wireshark and having the Velodyne driver connect to the replayed data? What does this show?

JWhitleyWork, 2026-11-16:
[written by AI]

Audit update (2026-11-16): The attached screenshot shows a VLP-16 PCAP launch configuration, but the referenced capture is not attached, so I could not check the missing angular sector. Static review confirms the VLP-16-specific unpack path remains in ROS1 `master`; I found no merged fix linked to this report. [#285](https://github.com/ros-drivers/velodyne/pull/285) proposes VLP-16 scan-separation changes but is still open and has not been checked against this capture. Please attach a shareable PCAP and exact replay/configuration details.

Source: [audited branch code/documentation](https://github.com/ros-drivers/velodyne/blob/29abd0e1361cb7f5eda451d2b51c35eeca45e0d5/velodyne_pointcloud/src/lib/rawdata.cc#L330-L385).