ros-drivers/velodyne #311: multiple velodyne align problem

robotdevel, 2019-11-11:
underline is my launch file for operating two LiDAR

<launch>

  <group ns="lidar_left"> # IP 192.168.1.210, 2100

  
  <arg name="calibration" default="$(find velodyne_pointcloud)/params/VLP16db_Left.yaml"/>
  <arg name="device_ip" default="" />
  <arg name="frame_id" default="velodyne" />
  <arg name="max_range" default="130.0" />
  <arg name="min_range" default="0.3" />
  <arg name="pcap" default="" />
  <arg name="port" default="2100" />
  <arg name="read_fast" default="false" />
  <arg name="read_once" default="false" />
  <arg name="repeat_delay" default="0.0" />
  <arg name="rpm" default="600.0" />

  
  <include file="$(find velodyne_driver)/launch/nodelet_manager.launch">
    <arg name="device_ip" value="$(arg device_ip)"/>
    <arg name="frame_id" value="$(arg frame_id)"/>
    <arg name="model" value="VLP16"/>
    <arg name="pcap" value="$(arg pcap)"/>
    <arg name="port" value="$(arg port)"/>
    <arg name="read_fast" value="$(arg read_fast)"/>
    <arg name="read_once" value="$(arg read_once)"/>
    <arg name="repeat_delay" value="$(arg repeat_delay)"/>
    <arg name="rpm" value="$(arg rpm)"/> # default 600.0(10Hz)
  </include>

  
  <include file="$(find velodyne_pointcloud)/launch/cloud_nodelet.launch">
    <arg name="calibration" value="$(arg calibration)"/>
    <arg name="max_range" value="$(arg max_range)"/>
    <arg name="min_range" value="$(arg min_range)"/>
  </include>

</group>

  <group ns="lidar_right"> # IP 192.168.1.220 2200

  
  <arg name="calibration" default="$(find velodyne_pointcloud)/params/VLP16db_Right.yaml"/>
  <arg name="device_ip" default="192.168.1.220" />
  <arg name="frame_id" default="velodyne" />
  <arg name="max_range" default="130.0" />
  <arg name="min_range" default="0.3" />
  <arg name="pcap" default="" />
  <arg name="port" default="2368" />
  <arg name="read_fast" default="false" />
  <arg name="read_once" default="false" />
  <arg name="repeat_delay" default="0.0" />
  <arg name="rpm" default="600.0" />

  
  <include file="$(find velodyne_driver)/launch/nodelet_manager.launch">
    <arg name="device_ip" value="$(arg device_ip)"/>
    <arg name="frame_id" value="$(arg frame_id)"/>
    <arg name="model" value="VLP16"/>
    <arg name="pcap" value="$(arg pcap)"/>
    <arg name="port" value="$(arg port)"/>
    <arg name="read_fast" value="$(arg read_fast)"/>
    <arg name="read_once" value="$(arg read_once)"/>
    <arg name="repeat_delay" value="$(arg repeat_delay)"/>
    <arg name="rpm" value="$(arg rpm)"/> # default 600.0(10Hz)
  </include>

  
  <include file="$(find velodyne_pointcloud)/launch/cloud_nodelet.launch">
    <arg name="calibration" value="$(arg calibration)"/>
    <arg name="max_range" value="$(arg max_range)"/>
    <arg name="min_range" value="$(arg min_range)"/>
  </include>

</group>
</launch>

--------------------------------------------------------------------------

i tried to align two LiDAR based on manual method.(LiDAR Calibration)

but VLP16 could not be aligned when i do that

i fixed two LiDAR same X, Y position.

and only Z and yaw will be different.

interesting part, when i align near +x part, +x area is align but -X area is not align

but when i tried to align near -x part, +x area is not align.

this problem appeared because of LiDAR accuracy?

could you guess what is problem?

JWhitleyAStuff, 2019-11-20:
@robotdevel - Can you provide a screenshot of the effect that you're seeing?

JWhitleyWork, 2020-03-20:
Closing due to lack of response.

closed 2020-03-20 by JWhitleyWork