ros-drivers/velodyne #312: hi, I find there are two calibration '.yaml' file for VLP16 in velodyne_pointcloud/params. which is the correct file? and what is the difference between them? Thanks.

Zhang-Qinghui, 2019-11-11:
**Is your feature request related to a problem? Please describe.
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
question
**Describe the solution you'd like**
A clear and concise description of what you want to happen.
I want some explantion.
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

JWhitleyAStuff, 2019-11-11:
1. In the future, please fill out the questions in the template for new issues.
2. One is for the VLP-16 (VLP16db.yaml) and one is for the VLP-16 Hi-Res (VLP16_hires_db.yaml). The Hi-res version has tighter vertical spacing.

closed 2019-11-11 by JWhitleyAStuff

Zhang-Qinghui, 2019-11-12:
Thanks for your reply. I will use the template for new issues in the future.