ros-drivers/velodyne #313: question about organized point cloud

Zhang-Qinghui, 2019-11-12:
**Please complete the following information:**
 - OS and Version: Ubuntu 16.04
 - ROS Version: Kinetic
 - Built from Source 
 - Version:  commit 7208295

**Describe the bug**
hi, organized clouds could  be accessed in a matrix fashion and I have tried, but I have one thing don't understand. Does the point under the same index number in the ordered point cloud of different frames come from the same position in the actual space? thank you in advance.

**Expected behavior**
I want some explanation about the problem.

JWhitleyAStuff, 2019-11-20:
@Zhang-Qinghui - There is an inherent problem with the current driver's implementation regarding the size of a "scan" - see https://github.com/ros-drivers/velodyne/pull/285 for explanation. The organized point cloud feature can only organize the data that it is given and, since the driver is inconsistent on the size of a "scan," the precise location of a point within a scan is currently non-deterministic. If you would like to work on this PR to fix this issue, help would be welcome.

closed 2019-11-20 by JWhitleyAStuff

Zhang-Qinghui, 2019-11-21:
Thanks for your reply. I think it's because the rotation speed of the radar is unstable.This is a hardware problem.