ros-drivers/velodyne #317: Velodyne drivers don't work with namespaces

VorpalBlade, 2019-11-25:
**Please complete the following information:**
 - OS and Version: Ubuntu 16.04
 - ROS Version: Kinetic
 - Built from Source:
 - Version: commit 2f1dd6b5

**Describe the bug**
I get the following messages:

```
[ WARN] [1574664548.218742413]: Timings not supported for model 64E
[ WARN] [1574664548.218834219]: NO TIMING OFFSETS CALCULATED. ARE YOU USING A SUPPORTED VELODYNE SENSOR?
```

This is strange since I'm using a VLP-16.

Turns out the cause is the code in rawdata.cc (line 200-201):
```c++
    ros::NodeHandle nh_driver("/velodyne_nodelet_manager_driver");
    nh_driver.param("model", config_.model, std::string("64E"));
```

This assumes that you don't use namespaces for the nodes (which I do). It should use a relative node path.

**To Reproduce**
Steps to reproduce the behavior:
1. Use the following launch file (adjusting path to calibration file):
```xml
<?xml version="1.0"?>
<launch>
  <group ns="sensors/laser1">
    <include file="$(find velodyne_driver)/launch/nodelet_manager.launch">
      <arg name="model" value="VLP16" />
      <arg name="rpm" value="1200" />
    </include>
    <include file="$(find velodyne_pointcloud)/launch/cloud_nodelet.launch">
      <arg name="calibration" value="$(find your_package_with_calibration_files)/calibration/vlp16_calib.yaml" />
      <arg name="min_range" value="0.1" />
    </include>
    <include file="$(find velodyne_pointcloud)/launch/laserscan_nodelet.launch" />
  </group>
</launch>
```
2. Attach a VLP-16 to the system
3. Start node
4. See message in console

**Expected behavior**
The Velodyne nodes should support namespaces.

JWhitleyAStuff, 2019-11-25:
@VorpalBlade - This was a bug that I missed in a pull request several months ago. Thanks for pointing it out! Would you mind testing #318 to make sure that the problem is solved in your case?

VorpalBlade, 2019-11-25:
Yes, I should be able to test that tomorrow

JWhitleyAStuff, 2019-11-26:
Fixed by #318.

closed 2019-11-26 by JWhitleyAStuff