ros-drivers/velodyne #319: catkin_make  error

lee1996-cnm, 2019-11-29:
Building CXX object velodyne/velodyne_driver/src/lib/CMakeFiles/velodyne_input.dir/input.cc.o
In file included from /home/liwei/catvehicle_ws/src/velodyne/velodyne_driver/src/lib/input.cc:56:0:
/home/liwei/catvehicle_ws/src/velodyne/velodyne_driver/include/velodyne_driver/input.h:58:18: fatal error: pcap.h: 没有那个文件或目录
compilation terminated.
velodyne/velodyne_driver/src/lib/CMakeFiles/velodyne_input.dir/build.make:62: recipe for target 'velodyne/velodyne_driver/src/lib/CMakeFiles/velodyne_input.dir/input.cc.o' failed
make[2]: *** [velodyne/velodyne_driver/src/lib/CMakeFiles/velodyne_input.dir/input.cc.o] Error 1
CMakeFiles/Makefile2:5260: recipe for target 'velodyne/velodyne_driver/src/lib/CMakeFiles/velodyne_input.dir/all' failed
make[1]: *** [velodyne/velodyne_driver/src/lib/CMakeFiles/velodyne_input.dir/all] Error 2
make[1]: *** 正在等待未完成的任务....
[ 67%] Linking CXX shared library /home/liwei/catvehicle_ws/devel/lib/libvelodyne_laserscan.so
[ 67%] Built target velodyne_laserscan
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

JWhitleyAStuff, 2019-11-29:
You are missing the `pcap` library. You can either install this manually by installing `libpcap-dev` or using `rosdep` to install all dependencies automatically.

closed 2019-11-29 by JWhitleyAStuff

lee1996-cnm, 2019-12-03:
Thank you so much. I have solved the problem. It's exactly as you said.