ros-drivers/velodyne #321: Incomplete Calibration object initialization

andre-nguyen, 2019-12-05:
**Please complete the following information:**
 - OS and Version: Ubuntu 18, applies to all
 - ROS Version: Melodic, applies to all
 - Built from Source or Downloaded from Official Repository: Source
 - Version: lastest master but I think the bug was introduced in https://github.com/ros-drivers/velodyne/pull/194

**Describe the bug**
Part of the Calibration parsing code in calibration.cc is used to retag all the laser_rings using the vertical angle. This code runs on the laser_corrections vector but not on the map, thus laser_corrections_map has incomplete initialization. 

https://github.com/ros-drivers/velodyne/blob/2e4e58259ddcf5c40e678eaac957b12114a06f07/velodyne_pointcloud/src/lib/calibration.cc#L170-L195

However, looking through the code, the map is only ever used to emit a YAML of the calibration and does it by using a forward iterator. Thus, the code below could iterate over the vector and we could technically remove that unused variable.

https://github.com/ros-drivers/velodyne/blob/2e4e58259ddcf5c40e678eaac957b12114a06f07/velodyne_pointcloud/src/lib/calibration.cc#L240-L249

**To Reproduce**
Read code

**Expected behavior**
Not have redundant datastructures.

**Additional context**
--

JWhitleyAStuff, 2019-12-05:
Thanks, @andre-nguyen! Would you mind submitting a PR?

JWhitleyWork, 2026-11-16:
[written by AI]

Audit update (2026-11-16): Static review of ROS1 `master` and `melodic-devel` confirms calibration parsing fills both `laser_corrections` and `laser_corrections_map`, then assigns ring numbers only in the vector. The emitter iterates the map, but does not serialize `laser_ring`, so I could not establish that this stale field affects emitted YAML. The separate map/vector duplication remains, matching the issue's request to avoid redundant structures. ROS2 `ros2` uses only the vector. No runtime calibration test was performed.

Source: [audited branch code/documentation](https://github.com/ros-drivers/velodyne/blob/29abd0e1361cb7f5eda451d2b51c35eeca45e0d5/velodyne_pointcloud/src/lib/calibration.cc#L146-L195).