ros-drivers/velodyne #324: Skip to specific frame or POSIX time within .PCAP file

istewar1, 2020-01-09:
Hello,

I have a request to add an option to skip to a specific frame or POSIX time specified by the user when reading a historic PCAP file. The default can start at the first frame of the PCAP file.

I have large PCAP files containing over 12 hours of data and I wish to skip to predetermined times when reading each file, rather than loading the entire file and reading the packets from the beginning.

JWhitleyAStuff, 2020-01-09:
That sounds like an excellent feature! Would you like to create a pull request to add it?

JWhitleyWork, 2026-11-16:
[written by AI]

Audit update (2026-11-16): I checked the current ROS 2 `ros2` branch and the ROS 1 `master` branch. Both PCAP readers consume matching packets sequentially, and the ROS 2 driver documents no option to start at a frame or POSIX time. I found no resolving PR; the requested start-position option remains unimplemented in the inspected source.

Source: [audited branch code/documentation](https://github.com/ros-drivers/velodyne/blob/8cdc0f37c8d6df1614bdbc424dd09f40a3c29251/velodyne_driver/src/lib/input.cpp#L302-L317).