ros-drivers/velodyne #325: Timestamps for every point from VLP-16

anastasiia-kornilova, 2020-01-16:
Hi! I'm looking (and waiting) for the feature of getting points' timestamps in velodyne_pointcloud. I found some fresh commits on this feature in master -- https://github.com/ros-drivers/velodyne/commit/19f62d282c89604b222eda4aca18bb07ecb4d44a. When it will be released? Is it tested and I can use this driver built from the sources?

JWhitleyAStuff, 2020-01-22:
@anastasiia-kornilova - I am sorry that I haven't released in quite a while. There was a pretty big change a few months ago and I want to make sure we have the result of that change mitigated. See https://github.com/ros-drivers/velodyne/issues/287#issuecomment-576338008. Once this is resolved, I will create a new release. You can use the current code on `master`, which has been tested on real sensors.

closed 2020-01-22 by JWhitleyAStuff

anastasiia-kornilova, 2020-01-26:
@JWhitleyAStuff , ok, get it, thank you a lot!